luc-github/ESP3D #442: [Question] BIGTREETECH-TFT35-E3-V3.0 with SKR Mini E3 V1.2 in Marlin Mode

Markus1970, 2020-04-15:
Hi All,

I have above mentioned combination running. The TFT can operate in Touchscreen-Mode and LCD (Marlin) Mode.
When the  TFT is switched to Touchscreen mode everthing is working fine so far. I can conrtroll the Printer via Webinterface in ESP3D. But If the TFT is switched to LCD mode I cannot controol the printer.
Is this by desgin or is there any setting in ESP3D which needs to be changed to get it working in LCD mode?

Thanks

Markus

luc-github, 2020-04-15:
I am guessing your esp is on TFT and not on SKR Mini
In that case is no ESP3D problem I think,
when you are in LCD mode the TFT stop communication by serial and so the ESP board on TFT cannot communicate with printer anymore, 
if you want to use the TFT in LCD mode then  you need to hook the Esp  board serial to SKR serial to let them communicate

Markus1970, 2020-04-15:
okay that makes sense now for me. Many thanks for the explanation. You are right, the ESP is on the tft and not direct on the SKR Mini. To have it wokring while the TFT is in touchscreen mode is fine for me.

many thanks 

markus

closed 2020-04-15 by Markus1970