luc-github/ESP3D #443: [BUG] AP mode not starting on newly flashed ESP-01

nlovrens, 2020-04-16:
Hi. 
After flashing ESP-01 (4MB)  with ESP3D version 2.1, AP mode is not starting correctly.
I am using esp8266 version 2.5.2 board, tried also with newest 2.6.3. Arduino version 1.8.12
Tried with second ESP-01 board, same issue.
Tried to upload WiFiAccessPoint example on the same board and it works normally.
On the wifi scanner I can see that AP is starting for a brief moment and then goes down.

[image]
 
Debug from the console:

> 14:53:37.059 -> SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
14:53:40.184 -> mode : softAP(82:7d:3a:34:eb:d7)
14:53:40.184 -> add if1
14:53:40.184 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
14:53:40.184 -> bcn 100
14:53:40.285 -> wifi evt: 8
14:53:40.827 -> wifi evt: 7
14:53:40.963 -> wifi evt: 7
14:53:49.504 -> 
14:53:49.538 -> SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
14:53:52.625 -> mode : softAP(82:7d:3a:34:eb:d7)
14:53:52.625 -> add if1
14:53:52.625 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
14:53:52.625 -> bcn 100
14:53:52.693 -> wifi evt: 8
14:54:01.915 -> 
14:54:01.915 -> SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
14:54:05.043 -> mode : softAP(82:7d:3a:34:eb:d7)
14:54:05.043 -> add if1
14:54:05.043 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
14:54:05.043 -> bcn 100
14:54:05.111 -> wifi evt: 8
14:54:05.111 -> wifi evt: 7
14:54:05.213 -> wifi evt: 7
14:54:14.344 -> 
14:54:14.378 -> SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
14:54:17.500 -> mode : softAP(82:7d:3a:34:eb:d7)
14:54:17.500 -> add if1
14:54:17.500 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
14:54:17.500 -> bcn 100
14:54:17.603 -> wifi evt: 8
14:54:26.899 -> 
14:54:26.899 -> SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
14:54:30.023 -> mode : softAP(82:7d:3a:34:eb:d7)
14:54:30.023 -> add if1
14:54:30.023 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
14:54:30.023 -> bcn 100
14:54:30.125 -> wifi evt: 8

luc-github, 2020-04-16:
hi can you share the output of 
`[ESP800]`
`[ESP420]plain`

nlovrens, 2020-04-16:
I don't have any output on the serial console for those commands.

This are IDE settings:
[image]

luc-github, 2020-04-16:
do you use reworked ESP01 or original one ?

nlovrens, 2020-04-16:
original:
https://www.ebay.com/itm/ESP8266-ESP-01-Serial-WIFI-Wireless-Transceiver-Module-Send-Receive-LWIP-AP-STA/292992245997?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

luc-github, 2020-04-16:
ok so your settings are incorrect ESP01 blue has 512KB flash when ESP01 black has 1MB flash - so no 4MB -  about reset method I guess you use a specific usb to serial to be able to use nodemcu so keep it if it works on my side I have basic one - I need to push button myself

try with these settings :
[image]

nlovrens, 2020-04-16:
Yes, you are right. It works with 1MB. I even tried before flash size test sketch and it reported that settings are correct for 4MB, so that is why i had it at 4MB.
Never the less, thank you for your support.

closed 2020-04-16 by nlovrens

luc-github, 2020-04-16:
No problem - have fun