luc-github/ESP3D #444: [FEATURE REQUEST] Pull output to reset line of printer board to allow printer firmware update

jan-gerard, 2020-04-18:
I assume that the "ESP3D Firmware Update" function in the "ESP3D Settings" tab of the UI refers to updating the ESP3D (e.g. esp8266) firmware. However, the ESP3D system could also be used to update the printer firmware.
Often, updating printer firmware requires a full serial connection to the printer including a reset line, or updating is done using a USB connection (that has a reset pull function in the CH340 -or other model- USB2serial chip). 
I can however update the printer firmware using the ESP3D interface. I did that several times. The only tricky part is the timing of manually pressing the RESET button on the printer (Anet A6 in my case) just before starting the PRINTER firmware update (for example using Octoprint). 

So my request is: 
* Add a PRINTER firmware update function to ESP3D and implement it such that you pull one of the I/O lines of the ESP board just before uploading the binary file. 
And, indeed, the ESP3D hardware should then also connect this I/O line physically to the reset line of the printer controller. But that is often located on the same connector of the printer where the ESP3D board is mounted.

luc-github, 2020-04-18:
So you just need a button on UI to set a GPIO state - I am correct ?

jan-gerard, 2020-04-18:
Not completely. I was thinking about a way to update the printer firmware (not the esp firmware) over wifi/esp3d. And a prerequisite is that the ESP should be able to pull the rest I/O line of the printer. 
For example Octoprint can update the printer firmware, put it cannot perform the reset by itself. So I see two options in my case:
1. Have Octoprint send a command to ESP3D (not UI, but http command) to pull reset, and then start its firmware upload.
2. Add a printer firmware upload function to ESP3D, that pulls the reset I/O line itself. 
I think that option 2 is most elegant, since it involves only a single component, the ESP3D board. But option 1 is easier, since Octoprint can already handle a firmware upload. Although I guess that uploading firmware is basically a matter of sending a couple of binary commands and then parsing the (binary) firmware file itself.
Just having a "reset button" in ESP3D will still involve the tricky timing of the 'manual' reset-before-flash action.

luc-github, 2020-04-18:
the command to directly command the pin already exists https://github.com/luc-github/ESP3D/blob/2.1/docs/Commands.txt
Check [ESP201] command, it can be sent from serial/http/tcp

for printer FW I am not sure what you mean by upload, upload where ? ESP3D can only upload to it's own FileSystem any file but size is limited, but even there what to do ?
using M28/M29  GCODE to upload file on printer SD card which is limited to GCODE text  file only

if you have more details about this printer update and octopi FW update process  I would be interrested

jan-gerard, 2020-04-18:
Is there a default state for the I/O outputs when ESP3D starts? Or is it randomly determined?

luc-github, 2020-04-18:
ESP pins start floating unless the board have some pull up you need to put state manually

jan-gerard, 2020-04-18:
I'll go for the http://192.168.1.108/command?plain=[ESP201]P12 option then first.

luc-github, 2020-04-18:
ok 
but I sitll do notunderstand difference betwen
> 1- Have Octoprint send a command to ESP3D (not UI, but http command) to pull reset, and then start its firmware upload.
2 -Add a printer firmware upload function to ESP3D, that pulls the reset I/O line itself.

jan-gerard, 2020-04-18:
I have a question about the I/O mode setting. The page https://github.com/luc-github/ESP3D/blob/2.1/docs/Commands.txt states:

```
*Get/Set pin value
[ESP201]P<pin> V<value> [PULLUP=YES RAW=YES ANALOG=NO ANALOG_RANGE=255 CLEARCHANNELS=NO]pwd=<admin password>
if no V<value> get P<pin> value
if V<value> 0/1 set INPUT_PULLUP value, but for GPIO16 INPUT_PULLDOWN_16
GPIO1 and GPIO3 cannot be used as they are used for serial
if PULLUP=YES set input pull up, if not set input
if RAW=YES do not set pinmode just read value
```

How do I switch back to input state **without** pullup, after using the I/O pin as reset output? Is that simply `command?plain=[ESP201]P13 PULLUP=NO`? It seems that simply requesting the pin state with `command?plain=[ESP201]P13` does not switch to input mode and keeps reading what last sent with the V0 or V1 command. Same when sending PULLUP=NO, so it seems like that is only valid as a PULLUP=YES combination.

The point is that after resetting via this I/O pin, I want to return to open-ended, tri-state, or whatever you would call the high-impedance output mode of the pin, which is then connected in parallel with the printer's reset button. And I don't want to force it high or low, unless I reset the printer with the ESP3D. 

By the way, I made this connection, hardware wired pin 13 to the reset line of the printer, and by that, flashing the firmware works!

jan-gerard, 2020-04-18:
@luc-github To come back to your question: the difference is that for option 1) the printer firmware upload is handled by Octoprint (avrdude in fact), and for option 2) the firmware upload would be handled by ESP3d which then should take over the role of avrdude. But I think that is much more work, and I got it working now anyway (apart from not knowing how to return the pin of the esp8266 running ESP3D to input-mode).
Octoprint > OctoPrint-FirmwareUpdater > executing a system command "wget http://192.168.1.108/command?plain=[ESP201]P13%20V0;wget http://192.168.1.108/command?plain=[ESP201]P13%20V1;sleep .2" > OctoPrint-FirmwareUpdater using /usr/bin/avrdude to upload the printer firmware

jan-gerard, 2020-04-18:
I used now the PULLUP=NO and that seems to do the job. Also according to the ESP3D source code. 
`wget http://192.168.1.108/command?plain=[ESP201]P13%20V0;wget http://192.168.1.108/command?plain=[ESP201]P13%20V1;wget http://192.168.1.108/command?plain=[ESP201]P13%20PULLUP=NO;sleep .1`
By the way: the source code suggests that if I don't specify PULLUP=NO, it will default to PULLUP=YES. Is that intentionally?

luc-github, 2020-04-18:
you can do a macro file on esp that do all commands
```
* Read SPIFFS file and send each line to serial
[ESP700]<filename>
```
and put this between the 2 commands
```
* Delay command
[ESP290]<delay in ms>[pwd=<user password>]
```
Yes usually pullup is the one I most used so I put it by defaut as need one by default, but usually better to put all parameters so no surprise

Yes avrdude is used for some boards - I am not sure it can be really fully done on ESP but there is a POC here : https://translate.google.com/translate?hl=en&sl=es&tl=en&u=http%3A%2F%2Fwww.sistemasorp.es%2F2014%2F11%2F11%2Fprogramando-un-arduino-remotamente-con-el-modulo-esp8266%2F

jan-gerard, 2020-04-18:
I think it would be leaner if the ESP3D created a pulse itself on a configurable pin, instead of having a server send commands to do that to the ESP3D board. 
So my suggestion would then be to add a command like
`[ESP202]P<pin> V<value> T<value>` that sends a pulse to pin P during T milliseconds with a shape not(V)-V-not(V). So if V = 1, it initializes the output pin P with 0, then switches to 1, and after T ms swithces back to 0, after which it is switched back to input mode.
Would that be an idea?

luc-github, 2020-04-18:
the macro file already allow to do what you need.
so adding a new parameter and code new process is not leaner IMHO, ^_^

jan-gerard, 2020-04-18:
Hi @luc-github , Sorry, I missed your post, didn't update my browser window. The macro is a nice option indeed. Thanks.

luc-github, 2020-04-20:
@jan-gerard  Any update?

jan-gerard, 2020-04-21:
@luc-github Sort of. I made a small macro that does the reset. It contains the lines 
```
[ESP201]P13 V0
[ESP201]P13 V1
[ESP201]P13 PULLUP=NO
[ESP290]100
```
Of course, I have made the connection from GPIO13 to the reset line of the printer.

However, there is no ESP3D command to execute a macro file, I believe. And flashing the printer firmware is not just uploading a binary file. Easiest is to use AVRdude for that. So just using the `[ESP700]` command is not going to work. I still need a combination of the firmware uploader plugin of Octoprint, sending a command to ESP3D. And it doesn't really simplify things if I make a macro on EPS3D and call that (if possible), or send 4 consecutive commands to it using wget.

jan-gerard, 2020-04-21:
But let's close this topic. I think I have a workable solution, and it would only get easier if ESP3D would include AVRdude functionality, which seems not that straightforward after all.

closed 2020-04-21 by jan-gerard

luc-github, 2020-04-21:
@d-a-v just suggested the usage of this lib : https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP

What is your printer board ?

jan-gerard, 2020-04-21:
@luc-github @d-a-v I have an Anet3d v1-5 board.
This looks like a nice solution -to include this libraray in ESP3D and create an interface for it- albeit that the MOSI and MISO lines are already in use for the LCD interface, and only Rx/Tx are available for programming. Unfortunately.

sroselli, 2020-05-09:
if i have averdudes where are the instruction for a flash A8-L shield for my Anet a8 v1.5 ? When run i receive this information : 

  >>>: avrdude -u -c arduino -p m1284p -P COM6 -b 57600 -U flash:w:"C:\Users\Sergio\Desktop\Stampante 3D\ANET\Firmware Originale\A8-L_firmwareUpdate.hex":a 
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0xd6
avrdude.exe: stk500_getsync() attempt 2 of 10: not in sync: resp=0x6f
avrdude.exe: stk500_getsync() attempt 3 of 10: not in sync: resp=0x80
avrdude.exe: stk500_getsync() attempt 4 of 10: not in sync: resp=0xbf
avrdude.exe: stk500_getsync() attempt 5 of 10: not in sync: resp=0xae
avrdude.exe: stk500_getsync() attempt 6 of 10: not in sync: resp=0xd6
avrdude.exe: stk500_getsync() attempt 7 of 10: not in sync: resp=0x02
avrdude.exe: stk500_getsync() attempt 8 of 10: not in sync: resp=0x21
avrdude.exe: stk500_getsync() attempt 9 of 10: not in sync: resp=0xcc
avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0xd6

avrdude.exe done.  Thank you.