luc-github/ESP3D #446: Can you give a little more info as to how to fit the code on a esp01s (1M)

jimhiggs, 2020-04-20:
I read your sticky but I am afraid I dont understand your statement 

> go to your ESP8266 core directory to disable float support
> type python ./tools/boards.txt.py --nofloat --allgen
> be noted this may affect your others ESP sketchs if you disable float support

I have always used platformio and got the original version of esp3d as a .bin file from the SKR PRO marlin distribution. I wanted to try the latest to see if any issues are sorted before I raised them.

luc-github, 2020-04-20:
This tip was given by @d-a-v the clever esp8266 contributor but was for arduino IDE - I am not sure how to apply this in platformIO if it is even possible easily

d-a-v, 2020-04-20:
Apart from commenting these two lines, I can't tell a proper way to do this with PIO.
https://github.com/esp8266/Arduino/blob/ea1fdb210f84fbed421340a1341376e179e4addb/tools/platformio-build.py#L91L92

jimhiggs, 2020-04-20:
Thanks @d-a-v I will give that a try.

luc-github, 2020-04-22:
@jimhiggs any update ?

jimhiggs, 2020-04-22:
I couldn't find the platformio-build.py anywhere on my pc. I could go back to arduino just for this project.

d-a-v, 2020-04-23:
That would be a question that you can ask in the esp8266 specific place in
platformIO forums.

jimhiggs, 2020-04-24:
I signed up but didn't have a clue where to start to pose the question in the right place. Luckily I stumbled on the users/me/.platformio location and was able to find platformio-build.py in that directory structure. so when I get a chance I will try building the latest ESP3D with the suggested mods.

jimhiggs, 2020-04-24:
How will I know if it will fit? Will it be a failure at upload time?

luc-github, 2020-04-24:
what is final firmware size ?
you won't have any error even config do not fit

jimhiggs, 2020-04-24:
I  ran platformio build  of newly git cloned esp3d The firmware.bin file 496082 
I commented the two lines out of the platformio-build.py and ran platformio build. 
The new firmware.bin file was 480956kb 
Is this the size you mean?  Is that a big enough drop?
I fitted a new unused esp01s in the pro wifi socket. I then renamed  firmware.bin to esp3d.bin and placed it on the sd card and  put it in the pro and rebooted. it looked like it was flashing but no sign after of a esp3d AP. Just to be sure I copied the esp3d.bin file from bigtreetech pro firmware github and was able to set up this board. I could try updating the firmware using teh ESP3D_WEBUI.

d-a-v, 2020-04-24:
The size drop looks fair regarding what you modified. Congrats!

I suspect what you did afterwards. Firmwares usually don't go on sdcards (we use esptool for flash it from serial link) but maybe esp3d has a flashing method that takes firmware from sdcard. But anyway that would work if you previously had an esp3d version already flashed which doesn't seem to be the case since it is a brand new(?) unused esp01s.
I suggest you use `esptool` to flash your firmware.bin file, then boot on it.
@luc-github may have better insights.

jimhiggs, 2020-04-24:
@d-a-v Ok thanks. But as I said I used the esp3d.bin from the bigtreetech github site using this method and it flashed and worked??

d-a-v, 2020-04-24:
Well, it looks like I don't know specifics of esp3d software enough but only lower level layers :)

jimhiggs, 2020-04-24:
You probably know a hell of a lot more than me. I had a career in SW but am retired now and learning new stuff is really hard!

luc-github, 2020-04-24:
Ok back on line - the size looks fine and will fit the 1M flash BTT ESP8266
One think is the SPIFFS size in platformIO.ini you need to change it for fit the IMB ESP8266, default size is for 4MB flash

luc-github, 2020-04-24:
Using WebUI to update FW won't work with 1M flash if FW is over 400Kb not enough space
yours is  480956kb which is still too high, I am not familiar with the BTT flashing method as I do not have such board allowing to flash ESP from SD card

jimhiggs, 2020-04-24:
Can I disable some features to make it smaller? All i need is the ui from my PC to monitor and start/stop prints and webcam. The esp3d.bin from BTT is 420kb and as I say that works. I reduced spiffs and rebuilt and reflashed but still no AP showing on 19.168.0.

luc-github, 2020-04-24:
480kb will works also - no worry but not webupdate
I doubt the 420KB allows webupdate neither
you can disable feature but the one which is nice is the one which use some space : notifications feature
I use the full features on my side without issue 
do you have a serial 2 usb adapter to connect your esp to your PC ?

jimhiggs, 2020-04-24:
Thanks Luc, the spiffs reduction did work. When I flashed before It tried in the tft and it did not work there either so I did not try after the spiffs change. Just tried it in the TFT and its working and I can see all the new bits around notifications in the ESP3D tab so I am now running the latest. 

Yes I realise the skr pro problem is not ESP3D, I am going to change my marlin to include @Bob-the-Kuhn's pin config PR and see if that help. Thank you for being patient with me.

luc-github, 2020-04-24:
no problem happy you managed to flash your esp ^_^

jimhiggs, 2020-04-24:
Not only that, I applied @Bob-the-Kuhn's pin config PR and I can now successfully use the ESP01S and ESP3D on the Pro. Who needs a Rpi and Octoprint!

luc-github, 2020-04-24:
great happy it works now

jimhiggs, 2020-04-24:
I still have a issue that is probably not ESP3D but you can tell me if I should raise it as an issue.
After starting a print from the onboard SD card. 
1 the Temperature does not report the target temp (ie bed is heating and it says 45/0). Is that because it does not know it?
2 I cannot abort a print. I also tried M524 from esp3d command line. Is that to do with marlin emergency parser needing to be enabled (which I can not enable because its not supported on stm32)

luc-github, 2020-04-24:
depending of what heat command is used Printer do not answer / any new GCODE command until target temp is reached
the only abort command possible  is from onboard SD card , from tft I did not checked if they allow any command

jimhiggs, 2020-04-24:
I did start the print from the onboard sd card . Could not abort.  If I click abort during homing it gives Error:Printer halted. kill() called! and halts the print. If I wait until the bed is heating the commands M108,M108,M108,M112 are issued but it carries on heating until bed reaches set temp and then it reports Error:Printer halted. kill() called! Sounds like a marlin issue or perhaps skr pro?

luc-github, 2020-04-25:
no may be np propee command I will check

luc-github, 2020-04-25:
The problem is when heating Marlin does not accept any commands when heating

you need to enable emergency parser :
```
// Enable an emergency-command parser to intercept certain commands as they
// enter the serial receive buffer, so they cannot be blocked.
// Currently handles M108, M112, M410
// Does not work on boards using AT90USB (USBCON) processors!
//#define EMERGENCY_PARSER
```

luc-github, 2020-04-25:
@jimhiggs please give a try to latest WebUI 2.1b60, it now handle all broadcasted info for temperatures / positions / feed rate / flow rate

jimhiggs, 2020-04-25:
Will try it today, thanks. I am not able to enable emergency parser. It errors, says it is not supported for stm32. It is also required for filament runout sensors and advanced pause which I have added to my printer.

luc-github, 2020-04-25:
@jimhiggs I see `EMERGENCY_PARSER is not yet implemented for STM32. Disable EMERGENCY_PARSER to continue.`
You may ask for it on Marlin Repository

jimhiggs, 2020-04-25:
Yes will do thank you.

sl1pkn07, 2020-04-25:
you can also, if you have weld skills, switch the flash to W25Q64FV (8M)

[image]
[image]
[image]

idea taked from here https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/3#issuecomment-534896903

greetings

jimhiggs, 2020-05-01:
Sorry Luc, I have been printing from the TFT SD card because I can't upload a file to the onboard SD card from ESP3D. 
I just tested and I can confirm that when I start a print from the onboard sd I can now see the temperatures and positions in ESP3D.
I tried to upload a file first it complained about 8.3 filenames so I renamed it to test.gco. It failed and locked up the printer. When I reset the printer I could not see anything on the onboard sd card from the tft or ESP3D. I manually changed the filename to test.gcode and I was able to see it on the TFT and see and start the print from ESP3D I thought 8.3 was 3 digit file types?
I think you have already mentioned you are looking at issues with upload.

closed 2020-05-01 by jimhiggs