luc-github/ESP3D #448: [FEATURE REQUEST]<scrollbars M165 Ax By Cz>

MarcFerrer, 2020-04-20:
Aadd scrollbars to set real-time filament mix%.

It is also possible to lock one bar to change% of one bar and the other to change automatically.

This is the option you can make in the menu of marlin mixier-> mix.

It serves as a diamond extruder.

thanks

luc-github, 2020-04-20:
You need to give more explaination, can you elaborate what is the request ? 
what are these control and what are the expected behavior ? - I am not familiar filament mixer

MarcFerrer, 2020-04-20:
I made a video with the printer working

[https://youtu.be/aRNvUFeQ9tg](url)

The 3 motors can be run at the same time to mix the filaments. Just say % of each color with the M165 command.

For example:
filament / Engine 1: M165 A1 B0 C0
filament / Engine 2: M165 A0 B1 C0

The 3 motors at once: M165 A0.33 B0.33 C0.33;

These changes are made by clicking on the different buttons in the scripts.

it would be nice if the% is could vary, in real time.

For example: color 1 to 20%, color 2 to 20% color 3 to 60%, which would be
M165 A0,2 B0.2 C0.3

To be able to make artistic pieces in terms of colors

I thought one way to do this would be with scroll bars.

And for it to be effective, a color is could be set with a checkbox and when moving the second the third would calculate it automatically until it reaches 100%

I thought one way to do this would be with scroll bars.

The result, a profile of rose (by Sant Jordi in Catalonia ;-)

[https://youtu.be/t4jvDIh400o](url)

luc-github, 2020-04-20:
so would give something like this ?
[image]

the button will send the GCODE command(s)command according value 0.1 0.3 and 0.6
 * Marlin M165 A0.1 B0.3 C0.6 
 * Repetier :  (for each extruder a command) M163 S0 P0.1 M163 S1 P0.3 M163 S2 P0.6  -
 * Smoothieware: seems not supported 🤔 so need to hide

here only A can be fixed because 3 extruders

if 4 extruders 2 can be fixed, N extruders N-2 can be fixed, the last value is the calculation vs N-2 extruder value

I am correct ?

MarcFerrer, 2020-04-20:
yes, it is correct and fantastic.

the repiter did not know him.

The option to fix (N-2) extruders also seems correct

luc-github, 2020-04-21:
you mean instead of A B C label ? put a configurable label ?

MarcFerrer, 2020-04-21:
that it is possible to put the colors of the filaments

[image]

luc-github, 2020-04-21:
this is special control that is not standard so need to write it I guess, but should be possible

MarcFerrer, 2020-04-21:
[image]
changing the background of the div?

luc-github, 2020-04-21:
yes should be easier

luc-github, 2020-04-21:
what the step should be  to reach 1 ?
0.1/ 0.05 /0.001 ?

MarcFerrer, 2020-04-21:
I think 0.05 would be enough

luc-github, 2020-04-21:
it can also be a preference parameter for mor flexibility

MarcFerrer, 2020-04-21:
like this?

[image]

luc-github, 2020-04-21:
yes can be also 
and color will be defined in preferences 
[image]

MarcFerrer, 2020-04-21:
There is one thing we can keep in mind.

A printer can have 3 extruders or it can have 1 extruder with 3 filament inputs for mixing.

I say this because in preferences it would not be correct to set the number of extruders to 3

Power would be fine to define a new panel named mixer. (with Marlin you can still do more things like virtual tools) where to specify the filaments and colors

MarcFerrer, 2020-04-21:
I may think about:

A printer can have 3 extruders or it can have 1 extruder and 3 filament inputs for mixing.

He says this because in preferences it would not be correct to set the number of extruders to 3

It may be okay to define a new panel (Mixer) (with Marlin you can do more like vital tools) where you can define the number of filaments, colors, etc.,

luc-github, 2020-04-21:
you double posted I think

the check box` Mixer extruders`do the trick - if checked it will make controls to appear or not in UI and preferences
Number of extruders is still the same but temperature control is then unique

MarcFerrer, 2020-04-21:
if it is correct. But I was saying it is not to be confused with other possible configurations (which I do not have)

Examples:
4 different extruders (no mixer)

https://www.youtube.com/watch?v=bn4gWYOzHxQ

The post of kina500 I think it refers to this configuration: (but of 5 inputs, it does not mix)

[image]

MarcFerrer, 2020-04-21:
yes but we have to consider one thing with Marlin:

filament splitter (no color control):  only GCODE T0...
[https://marlinfw.org/docs/gcode/T001-T002.html](https://marlinfw.org/docs/gcode/T001-T002.html)

filament mixer (which display color control) 
can go with M165
[https://marlinfw.org/docs/gcode/M165.html](https://marlinfw.org/docs/gcode/M165.html)

and here T0 ... are used by virtual tools
https://marlinfw.org/docs/gcode/M166.html
(I need to investigate more)

MarcFerrer, 2020-04-21:
A virtual tool saves a color combination.

for example:
M163 S0 P0.6 ; first component at 0.6. (in the Marlin menu of the Mixer says components)
M163 S1 P0.2 ; second secont at 0.6
M163 S0 P0.2 ; third component at 0.6
M164 S0          ;save to the virtual tool 0 the mixture (T0)

M163 S0 P0.33 ; first component at 0.6. (in the Marlin menu of the Mixer says components)
M163 S1 P0.33 ; second secont at 0.6
M163 S0 P0.33 ; third component at 0.6
M164 S1            ;save to the virtual tool 1 the mixture (T1)

to activate a tool of these 2 defined

M166 T
T1 ; activem la tool 1

(we would get the same color result with  M165 A0.33 B0.33 C0.33)

but since we've defined virtual tools, Marlin gives us more options by working with them using the M166 gcode
[https://marlinfw.org/docs/gcode/M166.html](https://marlinfw.org/docs/gcode/M166.html)

(Marlin uses S0 sometimes as A.. and sometimes as T0...)

luc-github, 2020-04-22:
@MarcFerrer I have moved  feature request to webUI

closed 2020-04-22 by luc-github