luc-github/ESP3D #449: FR100% still displayed when verbose turned off

jimhiggs, 2020-04-24:
Not sure if this is a bug or not. After first connecting ESP3D to my printer  I see the positions and FR100% continuously scrolling in the console/terminal window. Clicking off verbose stops the position statements but the FR100% reports keep on coming which is frustrating because it scrolls off screen other stuff I am trying to monitor/read

luc-github, 2020-04-24:
what is your configuration when happen , how esp is connected ?
some context would help

luc-github, 2020-04-24:
I am guessing you use BTT TFT which poll and broadcast lot of data in recent versions
Web UI do not consider it is a verbose info - I have moved it to verbose info in latest web-ui 2.1 - check latest git

closed 2020-04-24 by luc-github

jimhiggs, 2020-04-24:
Thanks luc-github. I am very new to github and I tried for an hour to work out how to merge the webui  repository with the latest esp3d and couldn't get it to compile so got some learning to do before progressing.

luc-github, 2020-04-24:
read carefully the readme of each repository
ESP3D and ESP3D-WebUI, unless you do code modifications you do not need to 'merge' / 'compile' I rebuild the index.html.gz almost for each changes multi lang is at root of repository and splited languages are in splited language directories

jimhiggs, 2020-04-24:
Great thank you again.