luc-github/ESP3D #450: [Question] Receive continuous feedback from printer via ESP3D?

jan-gerard, 2020-04-28:
How could I get more data returned from a http call to my ESP3D board, to have for example an external program do some measurements on the bed leveling, or to tune the heater PID controller? 
Now, I get only the first some seconds or number of bytes as response, and that is it. But for example a G29 command starts bed leveling of my printer, and the report with the measurement data is skipped. Although I see it all in the terminal screen of the ESP3D web interface.
It is just that a http call does return prematurely before all data is received.
This might be some sort of timeout of my browser, of wget, Matlab, or whatever interface I use to send/receive over ip. 
Does anyone have suggestions? This might not have to do with ESP3D at all, but I figure there are more people that could use this information, if there is a solution.

luc-github, 2020-04-28:
all printer output go to websocket this is what you see in terminal screen of esp3d
if you use external program you better use telnet command not http to get all data

jan-gerard, 2020-04-28:
Telnet is only supported in v3.0, which is in Alpha stage. I'd rather use a production version. It has no UI, I read.

luc-github, 2020-04-28:
telnet is present since 1.0

jan-gerard, 2020-04-28:
OK. I could only find it in the V3.0 documentation:
```
* Get/Set Telnet state which can be ON, OFF, CLOSE
[ESP130]<state>pwd=<admin password>

* Get/Set Telnet port 
[ESP131]<port>pwd=<admin password>
```

Which port should I use? I tried with puTTY to port 23, but no success. Should I enable it before compilation?

luc-github, 2020-04-28:
what about the readme? it is a good start
data port is 8888 and it is configurable in webui

jan-gerard, 2020-05-01:
I got it functioning. Indeed this works (after tickling my firewall). Now I can run a sequence from Matlab that probes the printer-bed repetitively to measure the reproducibility and small topology. 
And I am working on some off-line heater PID tuning, by doing a step-response system identification. 
The trick was indeed the time-out. My code now keeps listening on responses until it receives the final response string of for example the bed-probing function. This may take quite a while, and probably therefore the web interface terminated before receiving the complete dataset. How the 'terminal' function in Octoprint and ESP3D handle that (over http) I don't know, probably an asynchronous AJAX socket.

closed 2020-05-01 by jan-gerard