luc-github/ESP3D #451: [Question]<Enter comprehensive title>

Vnaf, 2020-04-29:
What is your question ?
Not sure if this is the best place for the comment but there I go anyway...

I've been installing this firmware in a Wemos D1 ESP32, the one with the UNO form factor, I got it working but needed to change the wiring a bit.

My ender 3 Pro has a V1.1.5 "silent" board, which is slightly different from the one shown in the wiki pictures.

This board has an CH340G which is not the same IC from the shown in the wiki.

I wired Tx from this IC (Pin2) to my ESP32 board Tx (the pcb trace goes to ATMega Rx passing through a resistor, so it's fine)

Then I tried to wire Rx from my ESP32 to the CH340 Rx Pin (Pin3), the result was ESP32 board not getting info from the printer's board. I checked the trace as I didn't want to solder into the ATMega and I found the line also had a resistor on it, I soldered in this resistor, in the side connecting to ATMega and its working nicely now.

I can take some pictures if needed.

BTW, great work! I'm loving the new functions added to my printer, specially the file uploading and everything I can do without using the encoder on it.

luc-github, 2020-04-29:
@Vnaf you can add your board description and your wiring in wiki, if you need to upload image you can use this issue or the one dedicated to wiki : https://github.com/luc-github/ESP3D/issues/427

The wiki is open to everyone ^_^
Thank you 😸

luc-github, 2020-05-04:
@Vnaf I close issue feel free to update wiki and ask help if you need

closed 2020-05-04 by luc-github