luc-github/ESP3D #452: Using Sketch with USB-ttl getting "error compiling board generic ESP8266 module"

ped7284, 2020-04-29:
Day 4 on my attempt on adding ESP8266 to TFT35 v3

Trying to follow all the directions but I am sure I am missing something.  I think I have the libraries in the correct locations. TO be safe I have in the following location and the c:/program files...

[image]

what questions do you have for me to get me pointed in the right direction?? and thank you.

Arduino: 1.8.12 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+111 (191024), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from sketch\syncwebserver.h:1:0,

                 from sketch\esp3d.cpp:88:

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:47:1: error: 'ESP8266WebServerTemplate' does not name a type

 ESP8266WebServerTemplate<ServerType>::ESP8266WebServerTemplate(IPAddress addr, int port)

 ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:68:1: error: 'ESP8266WebServerTemplate' does not name a type

 ESP8266WebServerTemplate<ServerType>::ESP8266WebServerTemplate(int port)

 ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:89:1: error: 'ESP8266WebServerTemplate' does not name a type

 ESP8266WebServerTemplate<ServerType>::~ESP8266WebServerTemplate() {

 ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:102:30: error: expected initializer before '<' token

 void ESP8266WebServerTemplate<ServerType>::begin() {

                              ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:108:30: error: expected initializer before '<' token

 void ESP8266WebServerTemplate<ServerType>::begin(uint16_t port) {

                              ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:114:32: error: expected initializer before '<' token

 String ESP8266WebServerTemplate<ServerType>::_extractParam(String& authReq,const String& param,const char delimit) const {

                                ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:122:30: error: expected initializer before '<' token

 bool ESP8266WebServerTemplate<ServerType>::authenticate(const char * username, const char * password){

                              ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:160:30: error: expected initializer before '<' token

 bool ESP8266WebServerTemplate<ServerType>::authenticateDigest(const String& username, const String& H1)

                              ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:238:32: error: expected initializer before '<' token

 String ESP8266WebServerTemplate<ServerType>::_getRandomHexString() {

                                ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:248:30: error: expected initializer before '<' token

[... 1616 lines cut ...]
 bool ESP8266WebServerTemplate<ServerType>::_parseForm(ClientType& client, const String& boundary, uint32_t len){

                              ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/Parsing-impl.h:601:32: error: expected initializer before '<' token

 String ESP8266WebServerTemplate<ServerType>::urlDecode(const String& text)

                                ^

C:\Users\phill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer\src/Parsing-impl.h:633:30: error: expected initializer before '<' token

 bool ESP8266WebServerTemplate<ServerType>::_parseFormUploadAborted(){

                              ^

Multiple libraries were found for "WebSocketsServer.h"
 Used: C:\Users\phill\Documents\Arduino\libraries\arduinoWebSockets
 Not used: C:\Program Files (x86)\Arduino\libraries\arduinoWebSockets
exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

luc-github, 2020-04-29:
please fill the issue template - you seems having wrong installation

ped7284, 2020-04-29:
Ok switched to branch 3 and now able to download.  I am able to see the wifi, log on to 192.168.0.1 PW 12345678

This is the screen.  I try to upload index but i get stuck here.  step 7.3 of instructions.  I think I have the correct firmware on the esp8266

[image]

Using ESP8266-01. 
Flash real id:   0014605E
Flash real size: 1048576 bytes

Flash ide  size: 1048576 bytes
Flash ide speed: 40000000 Hz
Flash ide mode:  DIO
Flash Chip configuration ok.

luc-github, 2020-04-29:
did you read the readme of the branch 3.0 ?

ped7284, 2020-04-29:
OK.  back ver 2.1 and same issue with original error....

luc-github, 2020-05-04:
I guess all is ok now

closed 2020-05-04 by luc-github