luc-github/ESP3D #453: [Question] connect esp8266 (as serial) slave/node with ESP3D esp8266 master/hub: intercomunicate devices

sl1pkn07, 2020-04-29:
What is your question ?

Hi

my actual:

esp8266 with ESP3D (master/hub) connect to printer board. the communication is PC <-> ESP3D <-> printer

my want:

connect another esp8266 (slave/node) to external device (custom prusa MMU2), comunicate both esps as serial bridge for comunicate board to device. and use the master/hub esp with ESP3D as usual

tree:

~~~
PC <― ― ― ―> ESP3D <――――――――――――――――> Board
         (master/hub)
                ˄
                |
                └ ―  ―> esp8266 <―――> Device
                             (slave/node)

― ― ― Wifi
――― wired
~~~~

is possible doing this?

greetings

luc-github, 2020-04-29:
```
esp8266 <―――> Device
                             (slave/node) 
``` is like a PC connection / or mobile
but doing like this you need to manage the communication priority to be sure PC and device do not send commands at in same time as ESP3D will not trigger anything

sl1pkn07, 2020-04-29:
na. i drop my idea

seems need some more work which is out of my principal idea

sorry for the noise

closed 2020-04-29 by sl1pkn07