luc-github/ESP3D #454: [FEATURE REQUEST] Support for BTT SD Cloud V1.0

rafaljot, 2020-04-29:
I ordered this $10 module for test.

https://www.aliexpress.com/item/4000991485282.html

luc-github, 2020-04-29:
Do you know if there is any specs / documentation / source code ?
I saw this on FB but I have no idea how it works

luc-github, 2020-04-29:
@rafaljot I do not think I can do anything it is only connected to SD no serial communication with printer 
it is a WiFi SD

rafaljot, 2020-04-29:
I hope BTT will publish a schematic of this module soon, just like others, like SKRs (https://github.com/bigtreetech ) . Even if not, it is very simple.

I don't think it is only remote-SD. We can use USB to connect to printer. It looks like there is ch340 chip on board.

luc-github, 2020-04-29:
the USB is to flash the FW - I do not see how the usb will be able to communicate with printer - none of them are usb host

rafaljot, 2020-04-29:
I think it is something similar:
[image]

https://3dtoday.ru/blogs/massaraksh7/sd-karta-s-wifi-na-esp8266-dlya-3d-printera-plug-and-play?fbclid=IwAR3Re1DvyVIkiezvatVK9iViC1FurmdtnGOtBKMQJc4QwWB763S28cFTv2s

luc-github, 2020-04-29:
yes I know this one, the project is not really new  - ESP3D do not support this way to hook the SD card , it is not reliable for me to control properly the sharing
you see usb is for PC - so BTT is a wifi SD not a printer controler

luc-github, 2020-05-01:
I close topic as no info about hardware/fw feel free to re-open if any update on it

luc-github, 2020-07-31:
I got one device and I will received also the FYSECT version 
I have flashed the BTT device with FYSECT FW and it is working so it means they 100% pin compatible

Now I have enough info I can work on this 
Also there is a webserver hooked version which is not sticked to espcore 2.4 https://github.com/d-a-v/ESPWebDAV/tree/v3 
so it should be funny to play with and do porting :

luc-github, 2020-07-31:
put this on BTT TFT 3.5 is not working well:
 * if connected when boot start it says `workibng on it` then  `Marlin is reading from SD Card`, if refresh windows, it is fine and list all SD content
 * if check sd content from TFT then SD is no more accessible it says  `workibng on it` then  `Marlin is reading from SD Card` which is normal but after refresh it says `This folder is empty` so no more Sd content is available
=> solution reset TFT then WiFi can access again... I am guessing the tft do not release the SD after query it unlike it is done at start, so BTT FW may need tuning ....

Bastlwastl84, 2020-11-13:
this would be very interesting if only uploading is working for the beginning and i could start print from the ESP-01 at BTT_TFT35_V3_E3 or directly from the Touchscreen. Tried your ESP-Firmware, but couldn't read from the SD-Card at the BTT_SD_Cloud at the user interface. the original-firmware from BTT is not working for me

luc-github, 2020-11-13:
no the esp3d does not support this hardware, original FW from BTT and the one from FYSETC is not really working for me as mentioned above

GilesBathgate, 2021-11-19:
@luc-github it should be possible to add usb-host support with the software driver https://github.com/cnlohr/espusb 
GPIO12 and GPIO13 look like they might be available for some patch wires.

luc-github, 2021-11-19:
this is not usb host, it is emulate block device like keyboard or mouse
why would you need that on sd wifi ? it already has usb connector on btt and fysetc one

GilesBathgate, 2021-11-20:
@luc-github Hmm, Yeah you are right, apparently I've googled esp8266 usb-host and got to the espusb page without checking the detail. I thought it was a software implementation of the shield as discussed here: https://www.hackster.io/139994/plug-any-usb-device-on-an-esp8266-e0ca8a

closed 2022-01-24 by luc-github