luc-github/ESP3D #455: What ESP module is recommended?

ModMike, 2020-05-02:
What is the best out of the box ESP module recommended for ESP3d as of today? I prefer 3.3v.

JuanKaMas, 2020-05-02:
i use NodeMcu V3 Lua and work fine

luc-github, 2020-05-03:
Any esp8266 board with at least 4MB flash 
for ESP32 I recommend 8MB if you want to use BT and WIFI if only WIFI 4MB is ok

NodeMCU is good one wemos is  also not bad

a use what you call `naked` esp-12 but you need to add some volage regulator or ou will get some crash
https://raw.githubusercontent.com/luc-github/ESP3D/master/images/Davinci/board.jpg

ModMike, 2020-05-03:
Luc,

Merci beaucoup pour l'aide! I saw this in your guide so I ordered it:

https://abra-electronics.com/wireless/wireless-wifi-en/wifi-esp-12f.html

1) How do I know how much memory is in these modules? Datasheets don't seem to say
2) Are all these modules programmable via FTDI board?

I am a few kilometres from a great electronics store that has just about everything arduino, robotics, and 3d printing in stock, although they are a little more expensive than China.

I have 2 different applications:

1) Add WiFi to resin printers that already have the WiFi header on board. Any plans to make ESP3D compatible with Resin printer firmware? As fas as I can see that would be very hard because they all seem to use the same closed source firmware and motherboard.

2) Use with a standard 3D printer and the Fystec Cheetah 1.1b board. They have a nice set of pins for the USB port right at the end. Unfortunately it has 3.3v pin, getting 5V is a little bit messier which is why I wanted a 3.3V module, but that looks to be impossible, or is it?

luc-github, 2020-05-03:
Esp8266 12 has 4MB flash
about resine printer I am not familliar but there no reason it won't work
if you do not use mks wifi module, others esp boards use 3.3v

ModMike, 2020-05-04:
> Any esp8266 board with at least 4MB flash
> for ESP32 I recommend 8MB if you want to use BT and WIFI if only WIFI 4MB is ok
>
> NodeMCU is good one wemos is also not bad
>
> a use what you call `naked` [...]

So based on your comments I bought a couple fo Wemos D1 Mini V3s to test with. Seems perfect in every respect, including 3.3v power.

ModMike, 2020-05-04:
The WEMOS board you have pictured looks very different than mine. My USB pins on the Fystec Cheetah are have 3.3V but how do I make sure the logic level is 3.3? Do I just measure it directly?

Don't want to blow anything hooking it up!

luc-github, 2020-05-04:
ESP8266 is only 3.3v on GPIO

what wemos picture are you refering to ?
there are several version of Wemos with also different flash size I have one with 4MB and one with 16MB

luc-github, 2020-05-23:
I think issue can be closed as you seems having your board working

closed 2020-05-23 by luc-github