luc-github/ESP3D #456: PlatformIO Seetings

ModMike, 2020-05-04:
Sorry about the dumb question, I haven't touched PlatformIO in 18 months.

Do I need to change the board ID to match my Wemos D1 Mini or leave as is?

Also do I need to specify the Global upload port for USB?

Thanks!

ModMike, 2020-05-04:
Never mind, bad USB cable!

luc-github, 2020-05-04:
Wemos D1 Mini is standard esp8266  so the default settings I have set should work

ModMike, 2020-05-04:
Thanks, found the link and sent donation.

I uploaded ESP3d 2.1 and the blue light doesn’t flash and I don’t see it on network anymore.

luc-github, 2020-05-04:
did you unplug replug after flash ?

ModMike, 2020-05-04:
Sorry, it took awhile and now it works perfect. Just need to upload Spiffs file. By the way, you show a logic level converter in the instructions but my board looks different. Do I still need it if my Fystec Cheetah board has a USB header with 3.3v?

luc-github, 2020-05-04:
the logic converter is only necessary if you printer board is 5V based, if it is 3.3v on GPIO there is not need

luc-github, 2020-05-04:
>Thanks, found the link and sent donation. 

@ModMike thank you, which link: paypal or sponsor ?

ModMike, 2020-05-04:
I did PayPal but can sponsor your amazing work if you like. Certainly worth it.

Sorry for another dumb question. I need to upload the spiffs file to get web UI right? Where do I find it?

Sorry I slept 2 hours and can't think straight.

luc-github, 2020-05-04:
Paypal ? -strange I should have been received it but do not see it - may be need time 
  
this one you need is https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

ModMike, 2020-05-04:
Hang on checking

luc-github, 2020-05-04:
Yes got it thank you ^_^

ModMike, 2020-05-04:
Ok everything up and running. Please excuse my stupidity, I feel like I was drugged today.

luc-github, 2020-05-04:
you need a rest ^_^

Ok my time to sleep then - have a good afternoon

ModMike, 2020-05-04:
Salut mon ami!

ModMike, 2020-05-04:
Plugged into board except need to figure out voltage issue. Right now powering each one from separate usb.

I don't see SD card but that's probably because of setup or 8.3 format. If I want to enable ESP3d-WebUI but want ESP3d to manage wifi do I just do the following in Marlin?:

`/#define WIFISUPPORT         // Marlin embedded WiFi managenent
#define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)

#if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT)
  #define WEBSUPPORT          // Start a webserver (which may include auto-discovery)
  #define OTASUPPORT          // Support over-the-air firmware updates
  #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host`

Why is this even needed if it talks to Marlin perfectly as is? It is already responding to commands with it off.

luc-github, 2020-05-04:
no Marlin settings you mention are for Marlin installed on ESP32 board and using ESP3DLib not ESP3D

check M20 is working this is what ESP3D WebUI use

luc-github, 2020-05-05:
@ModMike is it working now?

ModMike, 2020-05-05:
It’s working perfectly! I just need to get the voltage issue sorted out. Right now it only works if I give it per form it’s own USB connection, which is 5V.

I tried taking power from the 3.3V pin on the board but it does not work. I am hesitant to use the 5V. The board is also different than the one you have, but the name is the same.

ModMike, 2020-05-05:
Update: Got to work on 3.3v. This board is perfect!

luc-github, 2020-05-05:
That is weird here they write it can be done https://www.reddit.com/r/esp8266/comments/7ywmeg/powering_wemos_d1_mini_with_lifepo4_33v/

ModMike, 2020-05-05:
Check comment above your post.

luc-github, 2020-05-05:
ok so I guess issue can be closed
^_^

closed 2020-05-05 by luc-github

ModMike, 2020-05-05:
Yes sir, thank you for your patience. It's been a long week :-)