luc-github/ESP3D #457: Printer doesn't communicate with ESP3D

Arthur75012, 2020-05-05:
Hello,

I'm using ESP3D on a ESP01 connected to a gt2560 board with marlin on it.

The commands works fine, I can do home and control the temp but it seems that the informations goes only in one way : from the ESP3D to the printer. The other way doesn't work, the printer doesn't send any info... I can't see or upload (even with 8.3 file format) files in the SD card, the graph temp stay empty and nothing appear on the command pannel.

I try M115 or M20 and nothing happens
[image]

Arthur75012, 2020-05-05:
Also the printer panel stay empty too : 
[image]

luc-github, 2020-05-05:
looks like wiring issue check your TX on your printer  and RX on your ESP 
as you did not fill the issue template which give informations about your situation - I cannot  say more

Arthur75012, 2020-05-05:
The wire seems to be ok y checked the continuity between the esp TX, RX and the FTDI pin on the board (the board doesn't have TX and RX pin). I even try to invert the TX RX cable but in this case nothing works...

I can't really understand how it can work fine from the ESP3D to the printer but not in the other way. If the wire were bad nothing would work right ?

Sorry i'm really a beginer so what info do you need ?

luc-github, 2020-05-05:
when you open bug issue there is a template of info to fill.

* esp tx->printer rx = esp send commands
this part looks ok per your feedback

* printer tx -> esp rx = printer answer esp
this part seems not working

it is wire / connection  issue
be sure doing this if your serial is shared with usb that usb is not connected

Arthur75012, 2020-05-05:
Ok I effectively had a bad wiring 

Everything works fine thank you !

luc-github, 2020-05-05:
ok great, have fun :smile_cat:

closed 2020-05-05 by luc-github