luc-github/ESP3D #458: [BUG] 8.3 ERROR

EPDCenter, 2020-05-09:
[image]

I want to upload a file.gco but it always says it, i can read the sd files but i cant upload

EPDCenter, 2020-05-09:
[image]

Same message :(

luc-github, 2020-05-09:
this is not a bug you need to use 8.3 name format because this is the only one supported by Marlin
8 chars for name + 3 chars for extension

closed 2020-05-09 by luc-github

EPDCenter, 2020-05-09:
[image]

Not Work :S

luc-github, 2020-05-09:
I do not see the error message - `so what is not work?`

EPDCenter, 2020-05-09:
When I try to also upload that file, the error that I have put above it all appears, I have also tried for example ARMSS112.gco and it does not work either, it always gives me the error of 8.3

luc-github, 2020-05-09:
fill the issue template to get support without any information it is not possible to help

EPDCenter, 2020-05-09:
Solved... macOS was adding a hidden extension .gcode after edit the name, many thanks for your help, and sorry for it