luc-github/ESP3D #459: [Question] ESP32-DevKitC, WROOM-32D Can't connect to printer

jeffkyjin, 2020-05-10:
I am using ESP32-DevKitC, WROOM-32D for a testing ESP3D control my printer.
I cant connect to the printer, try to use the serial0, serial2 port， both not works.
Can't not get any printer information from the Printer Configuration of ESP3D web page. 

Any one have some suggestion?
Thanks a lot.

luc-github, 2020-05-10:
if ESP cannot communicate with printer : 
*  your printer serial is not configured -> your printer FW
* your printer serial is used shared with USB and your usb is currently connected -> disconnect your usb and restart boards
* wiring is not correct -> check your wires and be sure RX ESP go to TX Printer and TX ESP go to RX printer
* Baud rate are not same between printer and ESP

hilsonp, 2020-05-22:
Hello Luc,

Merci beaucoup pour tout ce travail.
Je continue en anglais pour que tout le monde puisse en profiter.

I'm modifying my da Vinci 2.0A to be able to create PCB at home.
I replaced one extruder with a 5W laser. The PWM control is connected to the extruder FAN. The second extruder has been replaced by a motor to drill the PCB.

Printer: XYZ da Vinci 2.0A Duo (but my motherboard looks like your pictures of your 1.0 ?!?)
ESP: ESP32-DevKitC, WROOM-32D

I compiled the Repetier firmware and uploaded it to the printer.
[config.h.txt](https://github.com/luc-github/ESP3D/files/4668847/config.h.txt)
The changes to the config.h are:
#define DAVINCI 3 // "3" For DAVINCI 2.0 with 2 FAN
#define MODEL 1 // "1" for new generation (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1 //Repurpose the main Extruder cooling fan to be controlled VIA M106/M107
#define FEATURE_DECOUPLE_TEST 0 //All heating device MUST be disconnected
#define CASE_FILAMENT_SENSOR_DEFAULT_ON 0
#define CASE_TOP_SENSOR_DEFAULT_ON 0
#define CASE_SOUND_DEFAULT_ON 0
#define BAUDRATE 230400

I can control successfully the laser using Repetier Host/Server over the USB cable.

Next step was to setup a ESP3D. I could compile it and uploaded it onto my ESP32 board.
I can connect to it over wifi.

The physical connection between the ESP32 and the XYZ is I understood from
https://github.com/luc-github/ESP3D/wiki/Davinci-1.0-and-2.0-board#where-to-connect-nodemcu-v3-on-davinci-10a-board
That is:
- Right pin of UART-PC (J11) connected to the TX pin of my ESP-32 (through a 330R resistor).
- Left pin of UART-PC (J11) connected to the RX pin of my ESP-32.
- ESP-32 connected to 5V and GND.

When I go to the ESP3D Printer tab, it doesn't show anything.

I tried inverting RX/TX wires (this is the connexion shown in my pictures).

It is not clear for me how to check if serial is enabled on the printer and if it conflicts with the USB port.

It is also not 100% clear if the right pin of the UART-PC port is the TX of the printer or should be connected to the TX of the ESP32. Indeed, the TX of the printer should be connected to the RX of the ESP32. I deduced my cabling from the wires color on your pictures.

Your help will be greatly appreciated !

[image]
[image]
[image]

luc-github, 2020-05-22:
Davinvi board use due chip which has several uart so there is no conflict with USB
you need to enable the second serial to be used by esp by enabling the setting WIFI
https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L65

hilsonp, 2020-05-22:
So. I did the following:

1. Disconnected the ESP3D.
1. Rebuilt/uploaded the Repetier FW to the printer with the following settings in the configuration.h (compared to the stock one):
#define DAVINCI 3 // "3" For DAVINCI 2.0 with 2 FAN
#define MODEL 1 // "1" for new generation (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1 //Repurpose the main Extruder cooling fan to be controlled VIA M106/M107
#define FEATURE_DECOUPLE_TEST 0 //All heating device MUST be disconnected
#define ENABLE_WIFI 1
#define CASE_FILAMENT_SENSOR_DEFAULT_ON 0
#define CASE_TOP_SENSOR_DEFAULT_ON 0
#define CASE_SOUND_DEFAULT_ON 0
#define CASE_WIFI_DEFAULT_ON 1
#define BAUDRATE 230400

2. Rebooted the printer
3. Went to the printer menu Settings/Load Fail-Safe
4. Verified that the wifi was activated in the menu. Wifi is on and the baud rate can be changed.
5. Powered off the printer
6. Connected "UART-PC right pin" to 330R resistor to ESP32 TX pin.
7. Connected "UART-PC left square pin" to ESP32 RX pin.
8. Powered on the printer
9. Connected to ESP3D web interface and verified that the "ESP3D Settings/Network/Baud rate" was set to the same value as the one found in the printer settings menu.

It does not seem to connect. 

I tried:
- swapping rx and tx
- changing baud rates

I tried reconnecting using my laptop connected to the printer usb with Repetier Server running on my laptop. I saw that the printer would connect and disconnect every few seconds.
The link became stable only when I disabled the Wifi in the printer menu.
Is this something you understand ?

Can you please confirm that on this "da Vinci 2.0A", I should connect to the UART-PC port like this:
- "UART-PC right pin" to 330R resistor to ESP32 TX pin.
- "UART-PC left square pin" to ESP32 RX pin.

What else can I try ?
How can I debug this ? From the ESP3D web interface ? By using the serial monitor on the Arduino IDE, connected to the ESP32 over USB ?

Thank you in advance.

luc-github, 2020-05-22:
why put resistor to esp Tx?
resistor is to protect esp so should be on rx pin not tx
I do not have 2.0a Mine is 2.0 so I can't comment what is wrong on printer but if serial is shared with usb you must do your tests usb  disconnected

luc-github, 2020-05-22:
also esp need gnd from printer to communicate

hilsonp, 2020-05-22:
I did put the 330R to ESP's TX because it was shown like this here:
https://github.com/luc-github/ESP3D/raw/master/images/Davinci/davinciA-1.jpg

I took another ESP32 from my drawer. This second unit was a rev1.
It still did not work but from the ESP3D/Printer Configuration, when trying to get info from the printer, it was taking forever.
When swapping the tx/rx pins or disconnecting fully the esp32, it was immediate (but no settings received).

I therefore deduced that when it was taking forever, it was communicating somehow.
I rechecked the printer menu and saw that the baud settings was different.
After correcting this, I suddenly worked.

This is GREAT !!!

So, what I did is:
- I used a ESP32-DevKitC, WROOM-32D rev1.
- In configuration.h of the Repetier FW, I enabled WIFI_SERIAL.
- In the printer menu, I made sure wifi was on.
- In printer menu and in ESP3D settings, I made sure baud settings was the same.

I will probably put a 330R resistor on both TX and RX.

Thank you for your help.

hilsonp, 2020-05-22:
And, yes, 5V and GND were from te printer.

luc-github, 2020-05-22:
indeed the picture is strange

luc-github, 2020-05-23:
but Due is 3.3v so it should be ok even no resistor

luc-github, 2020-05-23:
I think issue can be closed

closed 2020-05-23 by luc-github