luc-github/ESP3D #461: [BUG] ESP12E not connecting

roccotocco, 2020-05-11:
Hello, i tried to connect an ESP12E to my chiorn trigorilla board.... i followed this guide https://github.com/luc-github/ESP3D/issues/281
anyway i'm not able to get printer connected......wire is ok..... maybe can be a firmware issue ? into ESP3D Settings i set as Target FW "Marlin"
Somebody had some or similar issue ?
Thanks

luc-github, 2020-05-11:
No it is not a FW issue - it is likely an hardware issue (wires upside down, bad connection, etc), or a setting issue (different baud rate between printer and esp)

roccotocco, 2020-05-11:
thanks..... my printer is a anycubic chiron, maybe i choosed wrong firmware? its ok Marlin?
Thanks

luc-github, 2020-05-11:
I do not know this printer but using terminal with gcode should work, M115, M105, etc...

closed 2020-05-31 by luc-github