luc-github/ESP3D #462: Direct SD Card support is not opensourced yet?

DiegoGiovany, 2020-05-11:
Hi, long time ago do you said it will be open sourced.... I have a tinyfab cpu for tiertime printer, the firmware comes preinstalled and is awesome, but, some bugs don't allow reliable stream printing over wifi... the problem is, the only way to upgrade marlin firmware is pasting the firmware.bin in sdcard, and it only works on original firmware, not on the one compiled from source, but, to get nice prints I need to use the last code compiled from source.... Can you help opensourcing the sd over spi, or, maybe providing a updated version (pre compiled) for tinyfab cpu?  thanks a lot...

luc-github, 2020-05-11:
I do not know tinyfab board, sorry 
The original direct SD was done for Panucatt devices and it is custom FW for specific HW 
The direct SD access is already present in ESP3D 3.0 but there is no webUI yet, I am full time on it now and it is not compatible with Panucatt devices  hardware neither clones
If a 3.0 version is done for such devices it will be a custom version again.

DiegoGiovany, 2020-05-11:
hmmm, can I have a updated version for panucatt? maybe it works on my board.....  the version compiled from source, is almost perfect, the only missing thing is the hability  to send firmware.bin to the sdcard (the marlin upload only works for gcode).... thanks again !!!

luc-github, 2020-05-11:
I do not know what you call `updated version of panucatt` but you need to contact them if you need their FW as I do not provide custom FW on my own sorry

DiegoGiovany, 2020-05-11:
hmmm, I know it is not ideal, but if you not trust my usage, I can expose the esp3d web acess and maybe you can update it for me... If you can not do it... thanks anyway....

I have this cpu https://www.tinyfab.xyz/product-page/tinyfab-up-cetus-mk3-cpu-wifi

If you can, please update my board to a recent code that suports sdcard over spi (it comes from factory with it)..  Thanks again... and a will understand if you can not do it, but if you can, please help...thanks

luc-github, 2020-05-11:
the Panucatt FW is for the Panucatt hardware - I do not know what hardware your board has it can be a good clone or bad one so I cannot says if their FW will work on your board.

Also as I wrote I do not provide it as it is custom FW - sorry 

That said : `stream printing over wifi` is not good idea, you can loose connection / have slow response time that cause blob, so you cannot do fast print, etc...
better print from SD - always...

DiegoGiovany, 2020-05-11:
for stream print over wifi, with updated esp3d, marlin and octoprint, I can get 150mm/s clear prints, without fails until now,  the only real problem is, I can't update marlin without putting the printer apart, cause the sdcard is behind 9 screws.....   but ok, I understand.
I now my board can work with spi sdcard, because with the factory firmware, the option shows up and I can send firmware.bin to the sdcard, so... it works... I dont want to clone panucatt or nothing like this, just want to update marlin over wireless, but I understand, thanks by kindly answers and attention.

luc-github, 2020-05-23:
I think issue can be closed then

closed 2020-05-23 by luc-github