luc-github/ESP3D #464: [FEATURE REQUEST] Add BME280/BMP280 support

nanogennari, 2020-05-16:
BME280 and BMP280 sensors are much more reliable than DHT11 and DHT22, it would be nice to have support.

luc-github, 2020-05-17:
do you have a recommended library for esp32 and esp8266?

nanogennari, 2020-05-17:
The library Adafruit_BME280 (https://github.com/adafruit/Adafruit_BME280_Library) is in the Arduino repositories and it is quite simple to implement.

luc-github, 2020-05-17:
need also sensor library from adafruit...

luc-github, 2020-05-17:
this one looks more monolitic : https://github.com/MartinL1/BMP280_DEV

luc-github, 2020-05-17:
I need to think hw to do more generic sensor support, there is also https://github.com/luc-github/ESP3D/issues/432
Current support is focused on DHT but would be better to make more  generic for communication with webUI - As sensor support is not yet done in webUi 3.0 there is some time to think - TBD

luc-github, 2020-05-17:
Also I do not have any BMP280 device for the moment...

nanogennari, 2020-05-17:
If you need i can test and debug any code for the BME/BMP sensor. The idea of generic sensor support is nice, i even tried to alter your code to add it myself bot like you said the code is to focused on the DHT, anything i could do would be too jerryriged in place.

luc-github, 2020-08-25:
@nanogennari what sensor do you have BMP or BME ?
It seems I only got BMP sensors which do not have the humidity support

luc-github, 2020-08-25:
Implemented in FW - now I will update webUI to display it in notification area

closed 2020-08-25 by luc-github