luc-github/ESP3D #466: [BUG]<Enter comprehensive title>

KaktuMAN, 2020-05-23:
Hi

**To Reproduce**
Steps to reproduce the behavior:
No idea.

**Expected behavior**
I have a board arduino mega2560 WIFI with an esp8266, on the arduino I have a board RAMPS1.6. ESP3D is working but he can't connect to the printer. I can send commands to the ESP.

**Screenshots**
No one

**ESP3D Firmware:**
- ESP core version: [2.1] I think
- Version:[2.1.0.b39]
- Web UI Version:[2.1]  (I think)
- Wifi mode:[e.g. STA]  
- WebUI version: [e.g. 2.0b4] (I don't know)

**Target Firmware:**
 - Name: [e.g. Marlin]
 - Version [e.g. 2.0]

**Board used (please complete the following information):**
 - MCU: [e.g. ATMEGA2560WIFI] 
 - Name:[e.g. RAMPS 1.6] 
 - Flash size: [e.g. 4M: 2M/2M]

luc-github, 2020-05-23:
Dude, you have all version informations in webUI..., you flashed your board so you should know the versions you used (marlin / eps core) ...
 As no other reliable info  I guess your issue is :
> mega2560 WIFI
your Marlin configuration may not have a proper serial port configured for communication with embedded esp - I do not know ATMEGA2560WIFI] so I can't say more

KaktuMAN, 2020-05-23:
The problem was the baud rate in marlin I have to set the baud rate to 115200 in marlin (same as the board baud rate) And that's good

closed 2020-05-23 by KaktuMAN