luc-github/ESP3D #467: Single commands are OK. Gcode from ESP32 or upload to SD card fail [resend x]

hilsonp, 2020-05-23:
**Describe the bug**
After some time and with your help, I got my ESP32-DevKitC, WROOM-32D loaded with ESP3D and connected to my da Vinci XYZ 2.0A running Repetier FW.

From the ESP3D web interface, I can control the printer by sending single gcodes.
Nevertheless, as soon as I start something that loads the serial link, it quickly fails.

**To Reproduce - With playing gcode from ESP32**
Steps to reproduce the behavior:
1. Go to 'ESP3D / ESP3D Filesystem'
2. Click on 'Select files' and upload a gcode of about 20 lines.
3. Click on Close.
4. Assign this gcode to a macro on the control panel.
5. Launch the macro and follow the progress on the Commands panel.

It plays the first few lines of the gcode file but quickly fails (not always at the same line) and the Commands panel shows (for example) plenty lines like:
ok 5
ok 5
ok 5
ok 5
ok 5
ok 5
ok 5
ok 5
ok 5
ok 5
ok 5
Resend:6
ok
Resend:6
ok
Resend:6
ok
Resend:6
ok
Resend:6
ok
Resend:6
ok
Resend:6
ok

**To Reproduce - With uploading file to SD**
Steps to reproduce the behavior:
1. Go to 'SD Files' panel.
2. Click upload.
3. Select a gcode of around 20 lines.
4. Upload

It ends with "Error 0 :".

**Expected behavior**
Well. There must be something wrong with the serial connection but I could not find what to change.
I tried several baud rates, changes made on both sides.

**ESP3D Firmware:**
- ESP core version: ?
- Version: 2.1.0b39
- Web UI Version: 2.1.0b39  
- Wifi mode: Client Station
- WebUI version: 2.1.0b39
- Any change done in FW: No  

**Target Firmware:**
 - Name: Repetier for da Vinci
 - Version 0.92.10M

**Board used (please complete the following information):**
 - MCU: ESP32
 - Name: ESP32-DevKitC, WROOM-32D
 - Flash size: Can check but quite big enough

**Additional context**
The CPU clock is at 240MHz which may lead to https://github.com/espressif/arduino-esp32/issues/2004

But as I took the exact version asked in the wiki, I CAN4T CHECK IF THIS IS THE PROBLEM;

I may also try changing the SERIAL_RX_BUFFER_SIZE but I'm running out of idea.

This was the last thing I needed to proceed with my PCB etching machine :-(

Any help is appreciated.

luc-github, 2020-05-23:
the files  ESP3D are for few commands only / macro as they are sent without waiting any check so sending 20 lines may bring some overflow in printer or ESP, to send a full job GCODE must be on SD ESP3D is not a host for gcode yet

luc-github, 2020-05-23:
the resend  from SD means bad connection - use lower baud rate and be sure no noise on wire - need shorts lines to avoid any problem

hilsonp, 2020-05-23:
I'll see how to filter noise using capacitor this evening.

So do you suggest to put files on the printer SD by removing the SD card from the back of the printer, copying the files and  put the SD card back ?

luc-github, 2020-05-23:
I was refering to : 
>Go to 'ESP3D / ESP3D Filesystem'
Click on 'Select files' and upload a gcode of about 20 lines.

for SD upload it can take lot of time because protocole is very slow - on my DUO I use a wifi SD for upload

closed 2020-06-01 by luc-github