luc-github/ESP3D #468: [FEATURE REQUEST]<Support For Setting WiFi ON/OFF Command from Printer Display Menus

borland1, 2020-05-23:
Would like to be able to set WiFi transmitter state between ON or OFF from Marlin FW Printer Display Menus.

This would require:
a. ESP3D accept wired serial command: [ESP110]<state>
b. ESP3D to save State changes to EEPROM for subsequent power on cycles
c. Utilize Marlin FW macro and Custom Menu features to allow changing state of ESP3D (WiFi transmitter power ON/OFF).
d. Option for ESP3D to send normal M117 (Display Message) with "WiFi Switched OFF" instead of connected IP Address.

luc-github, 2020-05-23:
the 3.0 looks already meeting the request but the M117 message

luc-github, 2020-05-24:
is it really important to send the M117 WiFi Switched OFF at start ?

borland1, 2020-05-24:
An option to send M117 "WiFi State Set To OFF" is just desirable.
A message would not only provide feedback on state of ESP3D transmitter, but also a confirmation that ESP3D is being powered up and that wired serial is functioning.
Feedback to printer's operator would be similar to the current updating messaging when radio state is ON, with messages "Connecting.... ", "Connected.", "IP Address: 192.168.1.10".
I see that ESP3D v3.0 has set/get command [ESP110].  Marlin FW has an advanced  STARTUP_COMMANDS feature that allows sending g-code at power up. However because of the bootup sequence timing, it would be more difficult programming Marlin FW to: 
-pole ESP3D confirm bootup is completed,
-pole ESP3D to find transmitter state,
-then if it finds the radio transmitter state OFF, send it's own M117 display message reporting "WiFi State Set To OFF".

luc-github, 2020-05-24:
Ok adding the message if wifi off is not hard - I need to add message for ESP32  Bluetooth also in that case to be consistent

borland1, 2020-05-24:
To send ESP3D 3.0 commands from Marlin FW over wired serial, using g-code..... would this be correct?

M118 [ESP110] OFF pwd=12345678

luc-github, 2020-05-25:
yes, per Marlin documentation, `pwd=12345678` is only necessary if you enabled authentication

borland1, 2020-05-25:
For sending ESP3D 3.0 commands via hardwired serial, can the password parameter requirement be omitted or have a unique password for wired serial commands?  Unique that is, in having a hardwired serial password that can only be reassigned at source code level.

My thinking is that... If the admin password is later changed wirelessly with [ESP550], then Marlin FW would need to be updated to include the new ESP3D admin password.

luc-github, 2020-05-26:
No would would break security and having one password different

luc-github, 2020-05-26:
no removing password would break security purpose and having different password depending of input source bring complexity for edge case.
but 110 accept user and admin authentication  so use user password and do not change it ^_^

luc-github, 2020-05-31:
Implemented in 3.0 https://github.com/luc-github/ESP3D/commit/70a72ff8f39d1ee1824222ab60c490ba16f34cd6

closed 2020-05-31 by luc-github