luc-github/ESP3D #469: Using SD card from smoothiebord

sns5400, 2020-05-26:
After upload and change my rx/tx pins i can see rx and tx communication from printer to esp web interface. Temp information from head and bed is arriving i can move and home in X,Y,Z directions.
For the SD card i have a problem, must i do some special actions before i can read and write to the smoothiebord sd card ?

in the web interface it say:
[image]

luc-github, 2020-05-26:
nothing you set smoothiware as target is should be enough

what version of esp3d and  what version of web ui?

sns5400, 2020-05-26:
Do you mean i change in config.h
#define DEFAULT_PRIMARY_SD  2  --> 1
#define DEFAULT_SECONDARY_SD 1 --> 2

UI: 0.9.5 / FW: 2.1.0.b39

luc-github, 2020-05-26:
*   do not change these settings in config.h they are not for configuration purpose
*  use UI which match the FW : FW 2.1 based must use UI 2.1 based, not such old one

sns5400, 2020-05-27:
i thinks it work now, going to test
[image]

luc-github, 2020-05-27:
great I close ticket then

closed 2020-05-27 by luc-github