luc-github/ESP3D #472: Can not upload a file to SD card in a subdirectory

sns5400, 2020-05-28:
**Describe the bug**
I try to upload a file to a subdirectory on the SD card in my MKS smoothiebord.

I create a subdirectory with a capital letter in the name "Gcode". In the view i see only "gcode"

I try to upload a file to the SD / , this function is ok but very slow.
Remove the subdirectory "gcode" and recreate a subdir "gcode" = ok.
Change the view to the subdirectory by double click on the gcode subdirectory name = ok.
On top of the SD card view the path= /sd/gcode/  ok.
Then i try to upload a file with name "blokje.gco"  to my subdirectory "gcode" and i get a error:
"Upload failed"
"No SD card detected"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:**
- ESP core version: 2.0.1.b39
- Web UI Version:UI 2.1.b60  
- Wifi mode:STA fix ip adress  
- Any change done in FW 
 i change the RX and TX pin numbers to GPIO16 and GPIO17 of the ESP-WROOM_32 because have problems with default in config.h.

**Target Firmware:**
 - Name: MKS Smoothiebord
 - Version 1.3

**Board used (please complete the following information):**
 - MCU: ESP-WROOM-32
 - Bord: FireBeetle-ESP32 
 - Flash size: no info

luc-github, 2020-05-28:
please screenshot the error

sns5400, 2020-05-28:
Here are the screenshots:

[image]
[image]

luc-github, 2020-05-28:
Smoothieware answer if yout type `ls sd/gcode/`

sns5400, 2020-05-28:
[image]

sns5400, 2020-05-28:
[image]

luc-github, 2020-05-28:
As you can see smoothieware cannot open the subdirectory - I do not know why but it not linked to ESP3D

luc-github, 2020-05-28:
try with another name of directory many be - name `gcode` could be the problem

sns5400, 2020-05-28:
i try with the name test, same problem, is this a problem for arthur Wolf ?

sns5400, 2020-05-28:
when i give a command pwd it give me the path to my subdirectory
[image]

luc-github, 2020-05-29:
does `ls /sd/gcode` is ok?

sns5400, 2020-05-29:
Hello,
I copy on my pc 2 files on the sd card  1 in each directory
 in subdirectory "test" i put blok2.gco
in subdirectory "gcode" i put blok1.gco

Then i put the sd card back in my MKS smoothiebord and launch.
I have the follow result:
==  copy from screen  =========================================================
pwd
/
ls /sd/test/
Could not open directory /sd/test/
ls /sd/test
blok2.gco
ls /sd/gcode
blok1.gco

============================================================================
so this work normal

luc-github, 2020-05-29:
seems syntax changed since implemeted - I will add the fix for the path in webUI

sns5400, 2020-05-29:
[image]

i can even start the file for printing

luc-github, 2020-05-29:
yes the sd check use the `ls sd/gcode`  command to check it can start upload but now this syntax seems no more working - looking at doc it seems it is `relative path` syntax but was ok when I did several years ago - need to change the check command bay adding the `/` and will be back to normal I did not do such test since I implemented

luc-github, 2020-05-29:
I do not have a smoothieware available right now 
can you confirm this version :
[index.html.gz](https://github.com/luc-github/ESP3D/files/4700233/index.html.gz)
fix the problem ?

sns5400, 2020-05-29:
no good even worse, it double the sd/

File not found: /sd/sd/test/blok2.gco
ok
ls sd/
Could not open directory /sd/sd/

luc-github, 2020-05-29:
that is weird I do not double anything but just add "/" in from of path and be sure last char of path is not "/"

what actions are you doing ?
On my side I refresh SD:
[image]

then I created gcode sub directory:
[image]

then go to gcode sub directory  and upload small file:
[image]

I have also press print button:
[image]

This is what you described initially and It is working now

sns5400, 2020-05-29:
i test again afthernoon bizar

sns5400, 2020-05-30:
Hello, have tested this morning and have strange results. The double are disappeared bizarre anyway working with last version UI 2.1.b61 / FW 2.1.0.b39
When i ask ls in root there is no difference using / at end of command
When i do this for a subdirectory there is a difference
[image]
[image]

luc-github, 2020-05-30:
@sns5400 I did not change smoothieware commands, typing them will always give same result because they are smoothieware firmware commands

I changed webUI so please use webui to try to upload and list directory - do not use the terminal, the terminal commands were to debug what happened

sns5400, 2020-05-30:
ok sorry i have not understand your action correctly. Have just tested, the new web interface seems to work, the transfer speed is very low but that is an other problem.
[image]
[image]

even refresh after uploading the file in a sub directory.
Thank you

luc-github, 2020-05-30:
Great, thank you - I have updated webUI  git with the fix https://github.com/luc-github/ESP3D-WEBUI/commit/db672f36ae13fa95d7020f32c74ee932d87ea0fb

closed 2020-05-30 by luc-github