luc-github/ESP3D #473: [BUG]<Enter comprehensive title>

mike303625, 2020-05-29:
esp3d installed and running ok on node mcu8266, I have connected to mks sbase as recommended
in the install guide, making sure of  the tx and rx connections eg 8266 tx to sbase rx pin. 
In marlin I have defined second serial port as 0 (tried 1,3 as well and usb port is using -1) which is used for the normal tft screen.
When connected, printer does not respond and is not communicating with esp8266, does anyone have any suggestions as to what else to try?
My tx and rx are directly connected with no resistors
thanks in advance

mike303625, 2020-05-29:
my apologies this is a question not a bug

luc-github, 2020-05-30:
be sure baudrate is same on esp and printer, if you power mcu by usb be sure esp share gnd with mks

borland1, 2020-05-31:
If you have a FDTI serial adapter, you can use Arduino IDE's serial monitor to send G-code by RX/TX/Gnd serial port on printer board.  Once you confirm the printer board's serial port configured and working, connect your ESP8266.
After uploading the ESP3D 2.1 firmware, I immediately conneced it wirelessly to my PC in order to transfer the config files 404.htm, and favicon.ico files wirelessly.  I noticed by trial and error, that if my ESP8266 is too close (6 inches or less) to the PC, I was having difficulty connecting with very that strong signal.

borland1, 2020-05-31:
Any ESP3D documentation on the what the Marlin 2.0 firmware embedded "WiFi Support" provides?  I see it mentions ESP32 and ESP3D, but no mention of ESP8266.

luc-github, 2020-05-31:
@borland1 this is for Marlin FW installed on ESP32 based  board using basic WiFi or ESP3DLib (https://github.com/luc-github/ESP3DLib) not ESP3D which is for standalone board

There is no Printer board  ESP8266 based  and no Marlin FW for ESP8266 boards

mike303625, 2020-06-03:
just getting back to it, will try direct coms using arduino serial port try to find the uart config on the board! will advise if successful, the esp3d firmware appears to be working ok just not talking to sBase

luc-github, 2020-06-15:
I close issue - feel free to re-open if any update

closed 2020-06-15 by luc-github