luc-github/ESP3D #475: [Question] Wild idea: Cram a slicer into ESP3D

marciot, 2020-06-01:
Hello,

Since the start of the year, I have been a software developer at [SynDaver Labs](https://syndaver.com/) and a former developer at [LulzBot](https://www.lulzbot.com). For my new employer, I developed an entirely open-source web-based slicer based on the Cura engine:

[https://syndaverco.github.io/slicer](https://syndaverco.github.io/slicer)

This slicer is 100% client-side, so I had this idea of hosting it on an ESP module so that the user could point their web browser to a printer's IP address, drop their STL files onto the window, slice and hit "Print", without needing to install any software at all on their PC and without having to go through the intermediate step of downloading GCODE to an SD or USB drive (which is what they have to do now).

On further investigation, it seems like there may be some difficulties in doing this. The entire slicer weighs in at about 5MBs, and even zipped, 1.5 MBs. So this exceeds the storage on the ESP-01S modules most boards use, so a trivial hosting solution isn't possible on that chip.

That said, most printers have an SD slot on the motherboard. Does ESP3D have any capabilities of hosting web pages from a printer's onboard SD card? Even if it were just one zip file, that would be sufficient. The ESP-01S could serve out a page with a service worker that downloaded the zip, [unpacked it into the browser cache](https://serviceworke.rs/cache-from-zip.html) and use it to bootstrap the rest of the web-app.

Anyhow, just thinking out loud here. I would be very interested in learning ways in which Symple Slicer could be expanded to work with ESP3D!

-- Marcio

luc-github, 2020-06-01:
why not use online slicer?

luc-github, 2020-06-01:
I mean download the slicer code from another place than esp flash

luc-github, 2020-06-01:
sd support wont be possible if no direct connection with esp which is most of esp3d current configuration, but if there is an sd attached yes it is possible.
so add entry point in WebUi is may be the first approach.
that said  once sliced the gcode need to be uploaded to printer which can took long time to upload as 1MB is more than 20min in best case.
your solution is more for esp3d with direct access to sd like wifi backpack and Marlin on ESP32 board ans GRBL on ESP32 board

marciot, 2020-06-01:
> I mean download the slicer code from another place than esp flash

The idea was to make it so an internet connection isn't required. Military users come to mind. Also users who value their privacy and may also be skeptical of a hosted solution.

> that said once sliced the gcode need to be uploaded to printer which can took long time to upload as 1MB is more than 20min in best case.

Really? With a 250,000 bps serial connection, wouldn't it take just 30 seconds to transfer 1MB? Does the ESP have some limit on the max achievable baud rate?

A few more questions, if you don't mind:

The main page says ESP3D supports a WiFi to Serial bridge on port 8888.  This is a TCP socket, not a WebSocket, is that correct? I've just begun learning about WebSockets and my understanding is that JavaScript cannot connect to raw sockets. If so, how does the web component of ESP3D talk to the printer?

Let's take this one step further, if I did decide to host Symple Slicer from another server it would not be possible to have it talk to ESP3D; again, because of the host-only limitation on WebSockets, is that correct?

-- Marcio

luc-github, 2020-06-01:
yes really M28 protocol is not file transfer is command send one by one checked then save    to SD, super slow
there is a pinned issue explained this
2.1 use websocket to stream serial output from printer
websocket bridge is supported in 3.0  but this is not what you describe, you need webui support websocket communication from another host? but you said it is client side
for offline you may consider the slicer as browser extension, but it does not solve the post slice slow transfer to printer

luc-github, 2020-06-01:
according this test https://github.com/luc-github/ESP3D/issues/417#issuecomment-592511131
2.8MB = 35min

marciot, 2020-06-01:
> your solution is more for esp3d with direct access to sd like wifi backpack and Marlin on ESP32 board and GRBL on ESP32 board

I was not aware of the backpack. Thanks for telling me about that.

Do you know of any 3D printer board that use a ESP32 WROOM for WiFi rather than than the ESP8266? My understanding is that that module has up to 4MBs of Flash.

> ESP3D does not stream gcode yet

I'm planning to write a JavaScript streamer pretty soon, as one of the things on the Symple Slicer roadmap is making it into an Electron app so it can access a locally connected USB printer. This will make it more equivalent to the "native" Cura app.

> websocket bridge is supported in 3.0 but this is not what you describe, you need webui support websocket communication from another host?

Okay, let me clarify. As of today, Symple Slicer is a client-side web app. It has no ability to talk to printers either through serial or through TCP/IP.

For the next version (might be called Symple Slicer Desktop, or whatever), I plan to wrap it in an Electron wrapper. This will allow us to access local serial ports for tethered printing and flashing the printer. This will make it a viable replacement for Ultimaker's Cura.

The third possibility is for me to integrate Symple Slicer with ESP3D on a printer. I can see two ways that this could work. 1) either the printer serves out a web-based Symple Slicer itself via HTTP and then the JavaScript code running on the browser streams data using WebSockets to the printer or 2) the user downloads the Symple Slicer Desktop version (the one wrapped in Electron) and it connects to ESP3D's Wifi bridge using regular TCP/IP socket.

-- Marcio

luc-github, 2020-06-01:
wifi backpack is an daughter board using esp8266 that hook the lcd sd
most esp826612 but esp01 have 4MB flash, 2Mb for FW and web update 2MB for FS
ESP8266 can even reach 16MB (wemos d1-pro)

marciot, 2020-06-01:
> most esp8266 12E but esp01 have 4MB flash, 2Mb for FW and web update, then 2MB for FS
> ESP8266 can even reach 16MB (wemos d1-pro)

I see. So the ESP01 and ESP01S are the ones with little memory. I was looking at a BigTreeTech board which has sockets for those modules, so that's where I got the idea around the limitation. So thank you for clarifying that it isn't necessary to go to all the way to an ESP32, merely that it is something other than an ESP01 or ESP01S.

luc-github, 2020-06-01:
yes ESP01S has 1M flash, MKS boards use 4MB esp but they do not use ESP3D last time I have checked.

your solution looks really interresting hope you can achieve it

marciot, 2020-06-01:
> yes ESP01S has 1M flash, MKS boards use 4MB esp but they do not use ESP3D last time I have checked.

Ahhh, perfect! This is the board I need to get! Thank you! I imagine I could figure out how to get ESP32 running on it somehow.
 
> your solution looks really interresting hope you can achieve it

Thank you! And thank you for working on ESP3D. Even though I haven't used it (yet), I feel like it fits in well with what I have in mind... I just need to find the right parts to pull it all together!

Feel free to close down this ticket if you want. You've been a great help!

-- Marcio

luc-github, 2020-06-05:
Ok great - good luck in your investigation

closed 2020-06-05 by luc-github