luc-github/ESP3D #476: [BUG] Stack dump on ESP-01s on first boot

mattster98, 2020-06-08:
**Describe the bug**
On first boot after flashing, ESP-01s  serial output shows stack dump (attached).  Boot loops where it seems to try and initialize AP mode, but never gets there.

**To Reproduce**
Flashing v2.1 with basic 3.3v FTDI adapter, manually manipulating EN and GPIO0 pins to enable flashing.  Confirmed flash size (1MB) and set to 128KB FS.  Tried both 80mhz and 160mhz, and a few other settings with no noticeable change.  

Successfully flashed CheckFlashConfig during troubleshooting, so I'm pretty confident I have a working flashing setup.

Doing my best to follow the various step-by-step instructions.  Built on Arduino 1.8.12 (Windows Store) and tried several variants of the ESP8266 core (ones that will compile)

**Expected behavior**
Expected serial console to indicate AP mode & IP then I could connect.

**Screenshots**
N/A

**ESP3D Firmware:**
- ESP core version: 2.5.2 (also tried 2.7.1, 2.2.0 based on other bugs/guides)
- Version: 2.1
- Web UI Version: 2.1
- Wifi mode: AP
- WebUI version: 2.1
- Any change done in FW   - NONE

**Board used (please complete the following information):**
 - MCU: ESP-01s
 - Name: "AI-Cloud Inside"
 - Flash size: 1M

[debug_output.txt](https://github.com/luc-github/ESP3D/files/4747593/debug_output.txt)

luc-github, 2020-06-08:
1 - you must be in 160Mhz but this is not your current problem
2 - use generic esp8266 not other variant for ESP01
3 - Use 128KB for fs
4 - please screenshot all settings you use to flash in IDE
5 - raw  stack dump is useless for me - you must decode it using the decoder
 https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

I just flashed my ESP01 1M with these settings and latest git : FW version:2.1.1.b7

[image]

[image]
```
M117 ESP3D
M117 192.168.0.1
```
So issue is in flash settings or else where but need decoded stask to figure out

mattster98, 2020-06-08:
Thanks for the quick reply.  Just for "fun" I also tried 2.1 with Platformio and had similar results - just in case my Arduino+ESP8266 stack was broken.

Tried with your settings with 2.1.1.1 (didn't see a .zip of 2.1.1.b7) and latest ESP8266 core (2.7.1).  Arduino settings below.  No stack dump this time, but boot looping.  Will work on decoded stack trace now.

M117 ESP3D
M117 192.168.0.1

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
M117 ESP3D

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
M117 ESP3D

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
M117 ESP3D

[image]

mattster98, 2020-06-08:
Exception Decode below (got two different ones this time):

Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x4024571f
EXCVADDR: 0x40104bd2

Decoding stack results
0x40100380: pvPortMalloc(size_t, char const*, int) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\heap.cpp line 271
0x402243a4: ethernet_output at netif/ethernet.c line 312
0x401001b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 177
0x4022b85c: etharp_raw at core/ipv4/etharp.c line 1165
0x402280ee: netif_issue_reports at core/netif.c line 916
0x40228194: netif_set_addr_LWIP2 at core/netif.c line 717
0x40100c8c: calloc(size_t, size_t) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\umm_malloc\umm_malloc.cpp line 820
0x40100380: pvPortMalloc(size_t, char const*, int) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\heap.cpp line 271
0x40100394: pvPortZalloc(size_t, char const*, int) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\heap.cpp line 301
0x4020c4bd: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 432
0x40213274: __esp_yield() at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 119
0x4020c948: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 458
0x40205c9b: Esp3D::begin(unsigned short, unsigned short) at C:\Users\matt\AppData\Local\Temp\arduino_build_331071\sketch\esp3d.cpp line 169
0x40207a44: setup() at E:\Users\Matt Smith\Downloads\ESP3D-2.1\ESP3D-2.1\esp3d/esp3d.ino line 30
0x40213384: loop_wrapper() at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 194

Exception 4: Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register
PC: 0x400005cb
EXCVADDR: 0x00008000

Decoding stack results
0x4021377e: print_stack(uint32_t, uint32_t) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_postmortem.cpp line 211
0x40213957: __wrap_system_restart_local() at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_postmortem.cpp line 162
0x4021392d: __wrap_system_restart_local() at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_postmortem.cpp line 186
0x402139e0: delay_end(void*) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_wiring.cpp line 41
0x401001d9: esp_schedule() at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 125
0x402139ec: delay_end(void*) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_wiring.cpp line 44
0x40100380: pvPortMalloc(size_t, char const*, int) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\heap.cpp line 271
0x402243a4: ethernet_output at netif/ethernet.c line 312
0x401001b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 177
0x4022b85c: etharp_raw at core/ipv4/etharp.c line 1165
0x402280ee: netif_issue_reports at core/netif.c line 916
0x40228194: netif_set_addr_LWIP2 at core/netif.c line 717
0x40100c8c: calloc(size_t, size_t) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\umm_malloc\umm_malloc.cpp line 820
0x40100380: pvPortMalloc(size_t, char const*, int) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\heap.cpp line 271
0x40100394: pvPortZalloc(size_t, char const*, int) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\heap.cpp line 301
0x4020c4bd: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 432
0x40213274: __esp_yield() at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 119
0x4020c948: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 458
0x40205c9b: Esp3D::begin(unsigned short, unsigned short) at C:\Users\matt\AppData\Local\Temp\arduino_build_331071\sketch\esp3d.cpp line 169
0x40207a44: setup() at E:\Users\Matt Smith\Downloads\ESP3D-2.1\ESP3D-2.1\esp3d/esp3d.ino line 30
0x40213384: loop_wrapper() at C:\Users\matt\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\cores\esp8266\core_esp8266_main.cpp line 194

luc-github, 2020-06-08:
sorry it seems weird - seems not related to esp3d code

do you use the libraries present in repository ?

I suggest you to redo a clean install as the issue seems in core, remove all core elements and be sure not old versions are present
also try  use the same programmer as me

mattster98, 2020-06-08:
OK - I'm stumped.  I did have two versions of Arduino installed but uninstalled both, removed all of the Arduino* directories with libraries and hardware and just reinstalled from the Microsoft Store, latest ESP8266 core.  Copied the libraries from the repository into the new libraries folder and duplicated your Arduino settings.

Still just bootlooping with the occasional stack trace.  Feels like hardware but I've tried two identical units, one fresh out of the static bag.

Is there any other part of the hardware config (like bootloader or something?) that is a dependency here?  I haven't done anything to these devices other than use Arduino to try and flash them.

If no other ideas, feel free to close.  I'll move on without this for now.  Thanks for your help.

======
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
M117 ESP3D
M117 192.168.0.1

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
M117 ESP3D

luc-github, 2020-06-08:
hmm I remember I got issue with store version of ide, so I only use manual installation from  arduino website

I do not think it is hardware but installation
problem

mattster98, 2020-06-08:
Ok thanks.  I'll give that a try.  It wasn't as painful as I thought it
would be the first time.

mattster98, 2020-06-08:
Not enough power appears to be the issue.  I had suspected this possibility but figured there'd be stronger, clearer signals.  I got a different 3.3v source (another ESP-12 board with a large regulator) and guess what: it works now.

The weird thing is it didn't work when I plugged it into my SKR board (on USB power).  So, it must also have a weak 5v->3.3v regulator.  I *guess* that is solved when it is running off a real power supply.  Will try again once I'm done setting it up all the way.. change one thing at a time. :)

I did have to up the FS size to 192KB to get the index.html.gz to upload.  Other than that though, it seems to be working now.

luc-github, 2020-06-09:
144kb is enough if you use the limited langage version of index.html.gz

mattster98, 2020-06-09:
Ah ok.  I saw a note about selecting ewer languages but didn't see
how/where to do that.  Thanks.

It's still not coming up when on the SKR still even with 12v input so I'm
dead in the water either way.

luc-github, 2020-06-09:
what skr is it?

luc-github, 2020-06-09:
if it is a pro you need to use the latest Marlin GIT code as I am not sure this PR is in a release yet
https://github.com/MarlinFirmware/Marlin/pull/17679

mattster98, 2020-06-09:
1.4 Pro - thanks so much!  This is not my first printer, but my first attempt with this board and I'm definitely just barely stumbling my way along. :)

luc-github, 2020-06-09:
hope all will be fixed then

mattster98, 2020-06-09:
Looks like those add'l pins on the header on the SKR 1.4 Turbo aren't connected to anything, so I just soldered a jumper on the ESP01s to always enable it, so that part is sorted.

Now it doesn't look like the ESP is talking to the SKR.  Found a bug where the framework package for the lpc176x didn't have UART3 mapped to anything, so I fixed that, but nothing seemed to change.  Still working through that.

mattster98, 2020-06-09:
Aaaand it's because the SKR firmware wasn't actually updating even though it appeared to be.  All set now.  Thanks so much for your patience with me! :)

closed 2020-06-09 by mattster98

luc-github, 2020-06-10:
happy you succeed - have fun