luc-github/ESP3D #477: [Question]<Enter comprehensive title>

blaisebarrette, 2020-06-15:
I use ESP3D with GRBL_ESP32 and I would like to add the following line in the head section

`<meta name="apple-mobile-web-app-capable" content="yes">`

How do I go about recompressing it in .gz after my modification?

thanks!

luc-github, 2020-06-15:
you means esp3d-webui ?
this time it is here : https://github.com/luc-github/ESP3D-WEBUI/wiki/Compilation---Development

ESP3D FW is a FW for ESP board as daughter boards
GRBL_ESP32 is GRBL FW with some code of ESP3D for the WIFI features
ESP3D-WEBUI is the html page used as frontend by ESP3D and GRBL_ESP32, and ESP3DLib used for Marlin on ESP32 Boards

I hope it clarify ^_^

blaisebarrette, 2020-06-15:
It's a jungle out there! Hahaha!!! Ok then, yes I meant esp3D-webui.
The index.html.gz file found in the data folder that comes with the firmware GRBL_ESP32.

Can I uncompress the index.html file, add my few lines of code and simply reccmpress it back to it's .gz format? It's the recompressing bit I don't know how to do.

Thanks!

luc-github, 2020-06-15:
yes you can just use gzip to recompress , gzip is not zip

blaisebarrette, 2020-06-15:
OK! Thanks! I Kan manage that!
You've been very patient and helpful!
Many thanks!

- Blaise

closed 2020-06-15 by blaisebarrette

blaisebarrette, 2020-06-15:
It worked and I just wanted to say that it looks awesome as a webapp on an iPad!!!
If you haven't tried it, you should! It kick's ass!!!

luc-github, 2020-06-15:
I do not have ipad, can you share the screenshot?
so adding the meta and the link have big impact on ipad?

blaisebarrette, 2020-06-16:
Works with iPhone as well. Any iOS device. 
I made a short video so you ca see how it behaves.  

https://www.dropbox.com/sh/c6q9o0jbpqv208c/AAB6C015hJMZNf36A5QWFhV5a?dl=0

And here is a link where you will find all the info on what I did. It could’ve be simpler!

https://developer.apple.com/library/archive/documentation/AppleApplications/Reference/SafariWebContent/ConfiguringWebApplications/ConfiguringWebApplications.html

Tanks again!

luc-github, 2020-06-16:
Thank you for sharing - I will check that