luc-github/ESP3D #478: [FEATURE REQUEST]Add pre defined options in platformIO for flash size / type

luc-github, 2020-06-16:
Current platformIO.ini is for 
ESP8266 4MB Flash
ESP32 4MB Flash SPIFFS
Add entries for easily compile without modify platformIO.ini
for :
 * ESP01
 * ESP8266 16MB
 * ESP32 16MB Fat
 * TBD 
Is there a need for another target ?

DivingDuck, 2020-06-18:
Maybe ESP32-Cam?

luc-github, 2020-06-18:
@DivingDuck for PSRAM ?

DivingDuck, 2020-06-18:
Yes, but maybe I was misunderstanding and reading too fast (I was looking for a more specific setup for ESP32-Cam) .

luc-github, 2020-06-18:
you can do lot of setup for esp32cam and they most  need to be done in configuration.h
Here it is more about platformIO and hardware config
the platformIO.ini is not really intuitive if you do not know what to fill for hardware - in arduino ide all possibilities are in menu when in platformIO.ini you need to type the right flag which has not always easy to find documentation

luc-github, 2022-07-23:
using https://luc-github.github.io/#/config/features solve the issue |
So I close the ticket

closed 2022-07-23 by luc-github