luc-github/ESP3D #479: Integrating data into macro

JensHauser, 2020-07-03:
Hi,

I use an ESP32 with ESP3D and a working macro with XYZ edge finding.

Now I want to enhance it with potential work piece rotation (G68).
Based on the measured values I can calculate a rotation angle to shift the WCS fitting the work piece.

Inside a macro? Can I "get back" data from the CNC for calucations somehow?

Jens

luc-github, 2020-07-03:
currently macro just `send` commands and do not use / catch response
in 3.0 there is lua interpreter that would fit your need: send and handle response but not yet fully ready as miss user case

what you would do with the response of your command?

JensHauser, 2020-07-03:
My idea is to "ask" for the measured values of e.g. x1 and x2 (G38.2) at different locations of the work piece.
The difference should be calculated to a rotation angle of x axis at the current position, s.th. like G68 X0 Y0 R4.5

Maybe I´m thinking too complicated and there are other alternatives.

luc-github, 2020-07-03:
sorry to ask as it is not clear for me and I am not familiar with this command
who calculate the difference and where it is used?
should be automated ? do you need to add pauses in gcode to manualy do some calculation?

JensHauser, 2020-07-03:
Found a nice video explaining G68. He can do better than by written words :-)

DivingDuck, 2020-07-05:
@JensHauser, I like to ask you what controller board / firmware you are using for this?

JensHauser, 2020-07-05:
The board is a self-developed one based on ESP32 devkit v4 and the firmware is [GRBL_Esp32](https://github.com/bdring/Grbl_Esp32) and ESP3D master branch v2.1

Placing the XYZ finder on the work piece, editing the x1/x2 and y1/y2 distances in the ESP3D (so the edge finder can measure on existing workpiece and no beyond) and "go" would be nice :-)

@luc-github wrote about LUA, does this mean some scripting approach, which goes beyond GRBL sending and allows some interaction between script/user and GRBL?

DivingDuck, 2020-07-05:
Interesting. Didn't knew GRBL is supporting G68/G69. Seems I need to do an knowledge update. Thanks for the information.

JensHauser, 2020-07-05:
It doesn't yet, the functionality can also achieved by G38.2. If I just get back the X1/X2 and Y1/Y2 values, the script could calculate the corrective angle.

DivingDuck, 2020-07-05:
Ok, I got what you try to do. Thanks for your answer.

luc-github, 2020-07-05:
yes that is purpose of adding scripting language base is ready now need user case to add missing functions

JensHauser, 2020-07-05:
You have a volunteer for testing ;-)

luc-github, 2020-07-05:
well if you can share a situation with all actions that need to be done step by step it would be helpful for to see how to convert it in lua script

JensHauser, 2020-07-06:
Use case "XYZ edge finding with auto rotation"

Motivation
1. Overall target is
-    to switch from pure G code sender to an interactive scripting tool
-    allow getting back G code values (mostly what´s written in the "command" box of ESP3D, like coordinates)
-    feed them back into the calling script for calculations and assemble new G code operations before/during milling process
-    E.g. useful in or measurement operations ("where did the edge finder hit the work piece? -> point cloud")
2. A work piece can't be mounted exactly on work surface, so it´d be synchronized with 3D model.
   So there is a risk that milling is performed out of bounds
3. Finding work coordinate system (X, Y, Z) in order to start milling. In my case "G54"
4. Based on preliminary work (https://github.com/cncjs/cncjs/issues/118)

Prerequisites
1. Work piece is mounted tightly on work surface
2. Edge finder connected and placed on work piece
3. Work piece measures known
4. Milling cutter installed, a few millimeters above edge finder surface
5. Dimensions of edge finder are known to the program (height, width, depth, square landing area, )

Workflow
- Log into web interface 
- Start program "Auto rotate"
- Program asks for
  1. edge finder type (=various types with different sizes. I have an example...mine)
  2. milling cutter diameter [float]
  3. work piece measures X and Y [float]
  4. is milling cutter placed on expected area of edge finder? [boolean]
- Program starts 
- Measure x2/y2 and x1/y1
- Call function with (x2, y2, x1, y1) to calculate rotation angle
- Now 2 options
  1. Call G code to rotate coordinate system with "G68 ..." (not implemented in GRBL right now)
  2. In the G code generator, e.g. Autodesk Fusion 360, enter this angle to generate "rotated" G code
- Program stops

I have some more material for handling and calculations, but wanted to start with this in order to align the way we could go.

JensHauser, 2020-07-26:
Is it possible to show up a custom message in ESP3D master, e.g. "Please change from tool A to tool B now" ?

luc-github, 2020-07-26:
there is no such feature, you can only display text in terminal by usinh M118 command with Marlin like FW but grbl does not have such command as far I know

closed 2020-08-19 by stale[bot]