luc-github/ESP3D #480: [BUG]<Enter comprehensive title>

Patronics, 2020-07-04:
**Describe the bug**
ESP3D improperly handles access point with apostrophe (and presumably other special characters) in SSID.
Attempting to connect to a wifi network of the form `Name's Wi-Fi Network`, the "Join" checkbox button doesn't work, and attempting to paste in the name seems to lead to it being cut off as `Name` when reloading the view. (I haven't yet tried actually connecting, as I don't want the device to be unreachable with it's default access point until this is resolved. 

Furthermore, I attempted to escape the quotation mark with a leading backslash (`Name\'s Wi-Fi Network`), but that results in an even worse behavior upon reloading the view, as it now fails to load that configuration page, instead reporting `Failed:406 Wrong data`.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to freshly installed default ESP3D access point
2. Continue through the setup process until the Connecting-to-an-access-point step
3. Attempt to add an access point with a quotation mark in it's name. 
4. Notice that upon setting it and viewing later, the name is truncated to the portion before the `'`

5. attempt to resolve issue by escaping the apostrophe with a backslash
6. view error message upon saving changes and refreshing view

**Expected behavior**
ESP3D successfully connects to the access point

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:**
- ESP core version: unsure
- Version: 2.1.0.b39 ESP8266
- Web UI Version: 2.1b20  
- Wifi mode: Access Point

**Target Firmware:**
 - Name: Marlin
 - Version  2.0.x bugtest

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: ESP-01S]
 - Flash size: 1M: 1M/256k

Patronics, 2020-07-04:
An additional related bug, after the configuration file is confused/corrupted by the backslash, attempting to walk through the setup wizard again keeps the language as the only option onscreen, until you've clicked `Start Setup`, then `continue` 3 times, before reporting "Setup is finished", despite not offering any options aside from language.

luc-github, 2020-07-04:
does this code allow to connect with apostrope ?
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino

Patronics, 2020-07-04:
Thanks for the reply. Yes, it does successfully connect with the WiFiClientBasic.ino code you provided.

Patronics, 2020-07-04:
Also how can I reset the configuration data that's now inaccessible due to the "406 Wrong data" error? I reinstalled ESP3D and the filesystem, but it still seems to have the corrupted data stored somewhere, so I can't reconfigure or change the networking settings.

luc-github, 2020-07-05:
you can use https://github.com/luc-github/ESP3D/blob/2.1/docs/Commands.txt
`[ESP444]RESET` or configure manually the access pointwith direct commands

luc-github, 2020-07-10:
Fixed in web UI by : https://github.com/luc-github/ESP3D-WEBUI/commit/a42a7126cbfa3037cf0836dcd7a04339718ac1e8
Fixed in FW by : https://github.com/luc-github/ESP3D/commit/bb1007c4c441bb788bb93eba03605ae55907aca0

luc-github, 2020-07-10:
ESP3D 3.0 and WebUI 3.0 is also updated to fix same issues

closed 2020-07-10 by luc-github