luc-github/ESP3D #481: [BUG] Unable to Send V2.1 Commands Over Hardwired Serial TX/RX From Marlin 2.0.x Printer

borland1, 2020-07-06:
V2.1 does not respond to [ESPxxx] commands sent over hardwired serial from Marlin 2.0.x Printer.

Have successfully sent V2.1 [ESPxxx] commands wirelessly from ESP3D webUI, but have been unsuccessful doing same over hardwire serial sent from Marlin 2.0.x Printer by using M118 g-code.

From closed issue #468, this bug is apparently fixed in V3.0.  Can V2.1 be easily fixed in the interim while waiting for stable V3.0 release which is still scheduled for 'coming soon'.

**Expected behavior**
Would like to have ESP8266 respond to these commands sent from printer:  [ESP110] on, [ESP110] off, and [ESP110] restart.

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- WebUI version: [e.g. 2.0b4]
- Any change done in FW  Authentication Enabled 

**Target Firmware:**
 - Name: Marlin
 - Version 2.0.x

**Board used (please complete the following information):**
 - MCU: ESP8266 
 - Name: D1 Mini Pro
 - Flash size: 16Mb

luc-github, 2020-07-07:
the issue you linked is not related to M118
the commands need to be uppercase
ON OFF RESTART

borland1, 2020-07-07:
Actually the `Command.txt` file for V2.1 shows parameters in **UPPER CASE**, but if you look at your code, it only accepts **lower case.**

M118 g-code is used to ECHO text on HOST.  I used M118 to send [ESPxxx] commands wireless from webUI in browser and got confirmations.

luc-github, 2020-07-07:
Wow indeed - what did I do .... will fix that uppercase typo

but this won't explain why it does not work from M118
What is Marlin output for `M118 [ESP110]off`

borland1, 2020-07-07:
Marlin outputs:  `M118 [ESP110] off`  But nothing is echoed back to the WebUI and Marlin firmware does not sent returned text to LCD display.   
I never tried the `off` parameter wirelessly cause I'd need to re-flash the firmware again to over-write the radio EEPROM settings.
I was considering, but never tried, using a FDTI to send hardware serial directly from PC to ESP8266 radio.

luc-github, 2020-07-07:
the command off was just for the output look like
Marlin output is normaly `echo: [ESP110]off` 

I am out of home for still couple of days - I will when back home

luc-github, 2020-07-09:
@borland1 I have installed today the latest Marlin 2.0.5.4
I have setup on my SKR 1.3 the TFT 3.5 V0 with ESP01

I used : WEBUI: 2.1b61 / ESP3D: 2.1.1.b7

In Marlin

`#define CUSTOM_USER_MENUS`

```
#define USER_DESC_2 "M118 [ESP110]restart"
#define USER_GCODE_2 "M118 [ESP110]restart"
```

And it works ^_^ the ESP restart
 
But seems  M118 is not working when on SD file, I did not digged in Marlin code to check why - I have just tested and I seems not sending command

borland1, 2020-07-10:
That's good to hear.  I installed Marlin v2.0.5.4 firmware with "Custom Menu" feature, but have same fail result as before.

I have been using ESP3D v2.1.0,b39.  With branch 2.1 selected, I can view files content of version 2.1.1.b7, but the download to zip button provides older v2.1.0.b39.  Strange.

I will select branch 2.1-devt and download v2.1.1.b7 and give that a try.

borland1, 2020-07-10:
I tried with v2.1.1.b7 complied and uploaded with Arduino IDE 1.8.10.   Same result as with v2.1.0.b39.   With LCD menu restart or off commands, nothing is being echoed into wireless serial in webUI commands window.

Will try compile and upload from platformio over weekend.

luc-github, 2020-07-10:
Wrong action - I reopen
I think you need to check if Marlin send the M118 using an FTDI instead of ESP01 board
to see where the issue come from as on my side I cannot reproduce

borland1, 2020-07-11:
I used my `FTDI` serial adapter to troubleshoot the wired ports on both the ESP8266 and printer's control board.   I have Marlin v2.0.5.4 is configured for primary ( SERIAL_PORT ) and secondary serial port ( SERIAL_PORT_2 ) like this:

```
#define SERIAL_PORT   0
#define SERIAL_PORT_2 1
```
I had primary as the USB port and the secondary as wireless ( ESP3D ).  I also had configured LCD menu Custom Commands

```
  #define USER_DESC_1 "Set WiFi radio ON"
  #define USER_GCODE_1 "M118 [ESP110]on"

  #define USER_DESC_2 "Set WiFi radio RESTART"
  #define USER_GCODE_2 "M118 [ESP110]restart"

  #define USER_DESC_3 "Set WiFi radio OFF"
  #define USER_GCODE_3 "M118 [ESP110]off"
```

When using the Custom Commands, Marlin is correctly sending g-code commands, however they are being sent to the primary  serial port ( `SERIAL_PORT` ) instead of the secondary serial port ( `SERIAL_PORT_2` ).

So, I swapped the ports in the Marlin firmware, like this

```
#define SERIAL_PORT   1
#define SERIAL_PORT_2 0
```
 
Now Marlin sends the Custom Commands to ESP3D.   Problem now is sending the WiFi off command is not persistant with power off/on cycles.  Radio comes back on.  I thought it might be saved to the radio's EEPROM.

Now I need some help with saving the WiFi radio settings to EEPROM.  From the ESP3D webUI command window, sending command [ESP400] returns these EEPROM settings.
```
[ESP400]
{"EEPROM":[
{"F":"network","P":"112","T":"I","V":"250000","H":"Baud Rate","O":[{"9600":"9600"},{"19200":"19200"},{"38400":"38400"},{"57600":"57600"},{"115200":"115200"},{"230400":"230400"},{"250000":"250000"},{"500000":"500000"},{"921600":"921600"}]},
{"F":"network","P":"117","T":"B","V":"2","H":"Sleep Mode","O":[{"None":"0"},{"Light":"1"},{"Modem":"2"}]},
{"F":"network","P":"121","T":"I","V":"80","H":"Web Port","S":"65001","M":"1"},
{"F":"network","P":"125","T":"I","V":"8888","H":"Data Port","S":"65001","M":"1"},
{"F":"network","P":"130","T":"S","V":"MYESP","H":"Hostname" ,"S":"32", "M":"1"},
{"F":"network","P":"0","T":"B","V":"1","H":"Wifi mode","O":[{"AP":"1"},{"STA":"2"}]},
{"F":"network","P":"1","T":"S","V":"ESP3D","S":"32","H":"Station SSID","M":"1"},
{"F":"network","P":"34","T":"S","V":"********","S":"64","H":"Station Password","M":"0"},
{"F":"network","P":"116","T":"B","V":"2","H":"Station Network Mode","O":[{"11b":"1"},{"11g":"2"},{"11n":"3"}]},
{"F":"network","P":"99","T":"B","V":"1","H":"Station IP Mode","O":[{"DHCP":"1"},{"Static":"2"}]},
{"F":"network","P":"100","T":"A","V":"192.168.0.1","H":"Station Static IP"},
{"F":"network","P":"104","T":"A","V":"255.255.255.0","H":"Station Static Mask"},
{"F":"network","P":"108","T":"A","V":"192.168.0.1","H":"Station Static Gateway"},
{"F":"network","P":"218","T":"S","V":"ESP3D","S":"32","H":"AP SSID","M":"1"},
{"F":"network","P":"251","T":"S","V":"********","S":"64","H":"AP Password","M":"0"},
{"F":"network","P":"330","T":"B","V":"2","H":"AP Network Mode","O":[{"11b":"1"},{"11g":"2"}]},
{"F":"network","P":"120","T":"B","V":"1","H":"SSID Visible","O":[{"No":"0"},{"Yes":"1"}]},
{"F":"network","P":"118","T":"B","V":"11","H":"AP Channel","O":[{"1":"1"},{"2":"2"},{"3":"3"},{"4":"4"},{"5":"5"},{"6":"6"},{"7":"7"},{"8":"8"},{"9":"9"},{"10":"10"},{"11":"11"}]},
{"F":"network","P":"119","T":"B","V":"2","H":"Authentication","O":[{"Open":"0"},{"WPA":"2"},{"WPA2":"3"},{"WPA/WPA2":"4"}]},
{"F":"network","P":"329","T":"B","V":"2","H":"AP IP Mode","O":[{"DHCP":"1"},{"Static":"2"}]},
{"F":"network","P":"316","T":"A","V":"192.168.0.1","H":"AP Static IP"},
{"F":"network","P":"320","T":"A","V":"255.255.255.0","H":"AP Static Mask"},
{"F":"network","P":"324","T":"A","V":"192.168.0.1","H":"AP Static Gateway"},
{"F":"network","P":"168","T":"B","V":"0","H":"Notification","O":[{"None":"0"},{"Pushover":"1"},{"Email":"2"},{"Line":"3"}]},
{"F":"network","P":"332","T":"S","V":"********","S":"63","H":"Token 1","M":"0"},
{"F":"network","P":"396","T":"S","V":"********","S":"63","H":"Token 2","M":"0"},
{"F":"network","P":"855","T":"S","V":"??????????????????????????????????????????????????????????????","S":"127","H":"Notifications Settings","M":"0"},
{"F":"network","P":"170","T":"B","V":"255","H":"Auto notification","O":[{"No":"0"},{"Yes":"1"}]},
{"F":"printer","P":"461","T":"B","V":"0","H":"Target FW","O":[{"Repetier":"5"},{"Repetier for Davinci":"1"},{"Marlin":"2"},{"Marlin Kimbra":"3"},{"Smoothieware":"4"},{"Grbl":"6"},{"Unknown":"0"}]},
{"F":"printer","P":"129","T":"F","V":"0","H":"Output msg","O":[{"M117":"1"},{"Serial":"4"},{"Web Socket":"8"},{"TCP":"16"}]}
]}
```
I see that position `117` is `Sleep Mode`.  Not sure what value I should use for `Sleep Mode` = `WiFi off` ?  I see from above three options: `None`, `Light`, `Modem`.  Apparently `Modem` is my current setting.  What's the difference between `None` and `Light ` ?

Here's what I had in mind for the Marlin Custom Commands, but not sure about the V value? 

```
#define USER_DESC_1 "Set WiFi radio ON"
#define USER_GCODE_1 "M118 [ESP110]on" "\nM118 [ESP401]P=117 T=B V=2"

#define USER_DESC_2 "Set WiFi radio RESTART"
#define USER_GCODE_2 "M118 [ESP110]restart"

#define USER_DESC_3 "Set WiFi radio OFF"
#define USER_GCODE_3 "M118 [ESP110]off" "\nM118 [ESP401]P=117 T=B V=0"
```
Will try these later and edit this comment.

luc-github, 2020-07-11:
what is your printer board?

borland1, 2020-07-11:
It's a MKS_GEN_L_V2.  Uses an 8-bit ATMEGA2560.  V2 supports TMC motor driver UARTs. 

On this board, unfortunately AUX-1 is wired to USB; so I have remapped pins for Marlin's secondary serial port to the Mega2560's Serial1 pins occupied at by Z-min and Z-max connectors.  Moved Z-min limit switch to Y-max connector with pin override in Marlin's config.h file. 

If you look at the above EEPROM data, you'll see I have baud rate set at `250000`.   I looked at Expressif ESP8266 datasheet and it seems to says with 26Mhz external crystal, max baud is 74880.  However this does not seem to be the case. 

Do you remember the difference between Sleep Mode `None` and `Light` ?

borland1, 2020-07-11:
I found this..
```
wifi.setmode()

Configures the WiFi mode to use. NodeMCU can run in one of four WiFi modes:

Station mode (WIFI_STA), where the NodeMCU device joins an existing network
Access point (WIFI_AP) mode, where it creates its own network that others can join
Station + AP mode (WIFI_AP_STA), where it both creates its own network while at the same time being joined to another existing network
WiFi off (WIFI_OFF)
When using the combined Station + AP mode, the same channel will be used for both networks as the radio can only listen on a single channel.
```
So, I must be on wrong track.  Looks like I should use displacement `0` instead of displacement `117` and Value= `4`  ???

luc-github, 2020-07-11:
In 2.1 there is no permanent Wifi OFF so even you put 0 in position 0 of EEPROM it will not be OFF I have introduced  the support of permanent OFF only in 3.0
if the mode is AP in position 0 then it wil start in AP mode
if not it will start in STA mode

For Marlin good catch I missed the Pn parameter

AnHardt, 2020-07-11:
> Mega2560's Serial1 pins occupied at by Z-min and Z-max connectors. Moved Z-min limit switch to Y-max connector with pin overrides in Marlin's config.h file.

The endstop-pins are usually de-noised by more or less large capacitors. Not a that god choice for high baud serials.

borland1, 2020-07-11:
> > Mega2560's Serial1 pins occupied at by Z-min and Z-max connectors. Moved Z-min limit switch to Y-max connector with pin overrides in Marlin's config.h file.
> 
> The endstop-pins are usually de-noised by more or less large capacitors. Not a that god choice for high baud serials.

That is absolutely right AnHardt.  I had to remove one surface mounted pullup resistor and one de-coupling capacitors for each signal wire ( Z-min, Z-max ).

Here's pictures of the ESP8266 D1 Mini Pro assembly with custom PCB, TSX0102 voltage level translator, external WiFi antenna, and JST jumper cable.  The sandwiched assembly mounts inside printer's metal cabinet with velcro strips.  Antenna is inside, but also mounts inside with velcro, behind a plastic bezel for the OLED display.
[image]
[image]

closed 2020-07-11 by borland1