luc-github/ESP3D #485: How can I wipe the SPIFFS with a bin file?

rosellonsito, 2020-07-24:
I had the esp3d v2.1 and I saw the 3.0 version, so I updated first de webui to v3 to then upload the esp3d v3 and I cant do anything and if I try to downgrade I have the same SPIFFS, Is it possible to wipe it with a bin file? I upload the firmware to the mks wifi module with the sdcard. 
Any help? 
I can't downgrade the webui version because i dont have access to the main page
[image]

luc-github, 2020-07-24:
use the maintenance page : `http://<yourIP>?forcefallback=yes`

[image]

rosellonsito, 2020-07-24:
Oh my god! Awesome, I didn't know that.
Thank you very much!

closed 2020-07-24 by rosellonsito