luc-github/ESP3D #486: [BUG]SD card no refesh and uplod files

DrumClock, 2020-07-28:
Hi ,I have an ESP-01S WiFi module connected to the Touch Screen TFT 50 V3.0
However, I can't see any files on the SD card or upload to the card. The other panels Cntrols, Temperatures, Extruders and Commands work without problems.

[image]

[image]

luc-github, 2020-07-28:
1 - your FW version is very old and I guess you webUI is also very old as you did not fill the template information 
please update using latest git of each one

Also check if M21 then M20 commands list the SD content in console command - to see if SD is accessible

luc-github, 2020-07-28:
1 - please read the read me of the github
2 - M21 mount printer SD card and it says you have none, M20 list printer SD card and as you have none it does not show any
so it is normal
3 - if you want to list SD card of tft you must use `M20 SD:` instead but this only supported in latest git version of ESP3D and latest git version of ESP3D-WEBUI - so you need to update both
4 - the SD upload on TFT is not supported in BTT FW currently, you can only upload on Printer SD

DrumClock, 2020-07-28:
**3- Hi, `M20 SD: `command is functional in the console.**

**M20 SD:**
Begin file list
T1_Valec.gcode
T0_Valec.gcode
DUAL_Valec.gcode
++ servo center ++ .gcode
End file list
X: 160.00 Y: 135.00 Z: 21.20 E: 0.00 Count X: 12800 Y: 10800 Z: 42400
FR: 100%

**4- Unfortunately, my board does not have an SD card**
Do you know when support for SD FTF upload will be available?

luc-github, 2020-07-28:
3 - yes as you use TFT 50 - I was guessing you have latest FW which has my PR for TFT SD and TFT USB
but you still need to update your ESP with latest version of FW and webUI to get proper interface

4 - Sorry I have no idea - I do not think I will work on this by my self, also as ESP board is only connected by serial to TFT upload may be very slow compare to normal upload - I suggest you to put a WiFi SD if you need wifi upload - BTT do cheap ones but did not tested them - toshiba flash Air may be also an option but did not tested on TFT

closed 2020-07-28 by DrumClock

twinsenfan, 2020-09-19:
Hi there, 
I think I may have the same issue DrumClock had. I have a BTT SKR E3 Mini v2.0 with tft43 + ESP3D on a ESP01 board. The control board is running marlin bugfix-2.0.x.

All print control works as exptected but I am not able to upload. ESP M21 command returns "Card OK".

When you mentioned  the "SD upload on TFT is not supported in BTT FW currently", are you referring to the marlin firmware or the Tft firmware ?  

Thank you

luc-github, 2020-09-19:
upload on sd tft is not supported by tft fw, Printer SD upload is implemented

twinsenfan, 2020-09-19:
Can you let me know how to specify the printer SD to upload in the ESP3D UI ? Thank you.

luc-github, 2020-09-19:
you select SD, the BTT SD upload button is not visible when BTT SD is selected

twinsenfan, 2020-09-19:
Not sure I followed,  is "select SD" a button ? 

I see only button like "Progress" "Abort" "Upload" etc.

luc-github, 2020-09-19:
if you have upload button it goes to Printer SD