luc-github/ESP3D #487: [Question] Help interacting with ESP3D via JavaScript

marciot, 2020-07-28:
Hello,

I am currently working on a web based slicer (https://syndaverco.github.io/slicer) and I would like to have the capability of it printing to ESP3D. I envision there being a "Print to ESP3D network printer" option in the UI where it will either stream the file via a WebSocket directly to the serial port on the printer or upload a file to the SD card on the ESP3D and then start a print.

I'm not very familiar with how ESP3D works internally so I'm at a loss on how to even begin interfacing with it. Are there any examples I can look at on how to interact with ESP3D via JavaScript?

My first go is to try to connect to ESP3D via a WebSockets using the following code (my ESP3D is on 192.168.1.139 and the "Data" port is 8888. However, when I try this, it gives me a "Error during WebSocket handshake: net::ERR_INVALID_HTTP_RESPONSE" error.

```
<html>
    <head>
    </head>
    <body>
        <script>
            ws = new WebSocket("ws://192.168.1.139:8888"); 
        
            ws.onopen = function () {
                connection.send("M117 Testing");
            };
        
            ws.onmessage = function (e) {
                console.log("From Server:"+ e.data);
            };
        </script>
    </body>
</html>
```

luc-github, 2020-07-28:
connect to esp3d FW by websocket is only available in version 3.0,
 in version 2.x the websocket is only used to send serial output to webui as well as events, but not for communication.

the 8888 data port is for TCP data like telnet not websocket
Your code should work in 3.0 which still in alpha but you need to enable websocket https://github.com/luc-github/ESP3D/blob/3.0/esp3d/configuration.h#L58-L59

But to be honest I did not tested a lot this feature

marciot, 2020-07-28:
Okay. That's good to know, so it looks like with the current version of Panucatt's WiFi backpack I cannot use WebSockets.

So the other alternative is to simply upload a file and kick off a print. I came across a list of commands here but there is not documentation on how to send these commands:

https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands

The closes I came to finding what I needed was the following file:

https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/www/js/http.js

It appears to use GET and POST to communicate with ESP3D. Is this protocol documented anywhere?

luc-github, 2020-07-28:
the only doc is the code : https://github.com/luc-github/ESP3D-WEBUI/blob/84ca772a4d77d44e56e654149a1bd31b6e8d97ea/www/js/files.js

for panucatt devices you need to post the file to `/upload` 
and for reliability better provide file size in post also:
https://github.com/luc-github/ESP3D-WEBUI/blob/84ca772a4d77d44e56e654149a1bd31b6e8d97ea/www/js/files.js#L828-L835

luc-github, 2020-07-30:
seems websocket data was not configurable from webui interface - only by ESPXXX commands - I have updated FW and webui to support it properly

[image]

So far it is working on my side :
[image]

FYR the websocket terminal I have used is here : https://github.com/luc-github/ESP3D/blob/3.0/tools/wsterm.html

marciot, 2020-07-30:
@luc-github: Thank you so much for this! I had read another post by you where you mentioned the websocket terminal, but I had been unable to find it. Thank you for pointing it out! I will give it a try.

It looks like in my compiled version of ESP3D 3.0, there is not a "WebSocket Protocol". I'll have to look at the config files again, I must have missed something.

Anyhow, thank you so much for your help. I'm sorry for all the questions, but I hope that in the future I can return the favor and be of assistance in developing ESP3D in some way. It's a fantastic project!

marciot, 2020-07-30:
> for panucatt devices you need to post the file to `/upload`
> and for reliability better provide file size in post also:
> https://github.com/luc-github/ESP3D-WEBUI/blob/84ca772a4d77d44e56e654149a1bd31b6e8d97ea/www/js/files.js#L828-L835

Are the Panucatt devices considerably different than the base ESP3D? I'm just asking since you specifically call them out as being different. The Panucatt boards is one of the boards we are considering to add WiFi for our printers. You've warned that Panucatt made some modifications to enable the SD pass-through functionality, but we don't need that feature on our printer (our printer uses USB flashdrives rather than SD cards).

These are the boards we are considering:

- ESP-01S (lacks SD slot, so likely not very useful)
- ESP32-CAM (has camera connector)
- Olimex ESP32 Gateway (wired Ethernet + WiFi)
- Panucatt WiFi Puck

Anyhow, if you have any thoughts on these boards or recommendations for or against them, please let me know.

luc-github, 2020-07-30:
Panucatt wifi backpack / azteeg wifi / Viki WiFi - use private code esp3d made by me (and also owned) that use specific hardware present to allow to share SD card and so give easy access and fast upload to printer SD card
The WebUI is identical as 2.1

If you do not use SD and use USB yes you do not need  Panucatt board, but this means also you do not consider to upload file by wifi - I am correct ?
 
I have tested all 4 and each one is working with pro and cons but these depend on usage

Sorry I do not understand if you do not care on SD because you use USD disk, why put micro slot SD as first importance feature of board?

marciot, 2020-07-30:
> Panucatt wifi backpack / azteeg wifi / Viki WiFi - use private code esp3d made by me (and also owned) that use specific hardware present to allow to share SD card and so give easy access and fast upload to printer SD card

Just curious. Has that code been integrated into the main ESP3D codebase? If not, is there are reason to keep it separate?

> If you do not use SD and use USB yes you do not need Panucatt board, but this means also you do not consider to upload file by wifi - I am correct ?

The way I am considering it there would be two independent ways of printing:

- Plug a USB Flashdrive into the printer and use the LCD panel to initiate a print from Marlin.
- Use WiFi to upload GCODE to ESP3D board (the file gets stored in the microSD card on the ESP board) and initiate a print from ESP3D.

The microSD card would be installed at the factory and may not even be accessible to the user. The ESP3D would only have a serial connection to the main printer board for streaming prints via serial.

To make it easier for the user, I would like to see if I can modify ESP3D to accept a file upload via Javascript, save it to the SD card, print it and then delete the file when done. This would allow one-click printing from Symple Slicer, with ESP3D acting as a print spooler.

More advanced users could log in to the ESP3D WebUI for finer control over the printer, if they desired.

> Sorry I do not understand if you do not care on SD because you use USD disk, why put micro slot SD as first importance feature of board?

Because there needs to be someplace for the user to upload files via WiFi for printing. This cannot be the USB flash drive, for various reasons 1) I want the user to be able to upload files even if they do not have a USB flashdrive inserted 2) transferring files via serial to the main Marlin board is slow, 3) the electrical interface for the USB module is incompatible with the interface to SD cards, so something like a Panucatt board cannot be used to multiplex the USB interface.

luc-github, 2020-07-30:
> Just curious. Has that code been integrated into the main ESP3D codebase? If not, is there are reason to keep it separate?

Not yet - because the specific hardware was not available at this time and give some advantage to panucatt who supported me, it will be in 3.1 for sure, 3.0 has almost everything now but need to be stable enough to add it

>Use WiFi to upload GCODE to ESP3D board (the file gets stored in the microSD card on the ESP board) and initiate a print from ESP3D.

How ? ESP3D is not yet an host and if file is on ESP3D SD card, it is not accessible from printer so how printer can use it ? and so ESP3D cannot initate a print for a file that printer FW cannot see

ESP3D already accept upload using curl - no need more code I think 

Also being an host is not simple task, you need to have prioritary dedicated task to avoid any delay in sending command or you get some blob - pause, host also behave differently if Printer is using Marlin/Repetier/Smoothieware.

>The electrical interface for the USB module is incompatible with the interface to SD cards

What chip do you use for USB drive ?  CH376s? how is connected to printer board ?

marciot, 2020-07-30:
> How ? ESP3D is not yet an host and if file is on ESP3D SD card, it is not accessible from printer so how printer can use it ? and so ESP3D cannot initate a print for a file that printer FW cannot see

Ohhhh! That kind of throws a wrench into things. I had assumed ESP3D was a print host.

We do have a GCODE host in Symple Slicer, so we could still stream directly to the websocket, but that is a bit dicey since if the WiFi connection drops, the print fails.

It's possible I could port the host code into ESP3D, but that would be more of an advanced project than I had hoped for.

> What chip do you use for USB drive ? CH376s? how is connected to printer board ?

It uses a MAXIM MAX3421E, the same chip as used in the Arduino host shield. I took the open-source library for the Arduino host shield and integrated it into Marlin. The protocol is very low-level and CPU intensive -- it's somewhat of a miracle it works with Marlin at all.

luc-github, 2020-07-30:
>Ohhhh! That kind of throws a wrench into things. I had assumed ESP3D was a print host.

Yes detailed feasability study is very important when kick starting a project 😉 

>if the WiFi connection drops, the print fails

Yes printing from WiFi is very bad idea ^_^, I only print from SD

I have started the host code but need to redo it as not working as expected - also it will only work on ESP32 to be able to use wifi in same time without issue - forget esp8266 for such purpose IMHO

>MAX3421E

That is really overkill for just USB stick support!!!  I tested it before  and I decided on ESP32 to use the CH376S instead and this lib: https://github.com/djuseeq/Ch376msc

luc-github, 2020-07-30:
FYI you should read : 
https://github.com/luc-github/ESP3D/issues/239
and https://github.com/luc-github/ESP3D-WEBUI/issues/94#issuecomment-657225602

To get latest status

marciot, 2020-07-30:
> > Ohhhh! That kind of throws a wrench into things. I had assumed ESP3D was a print host.
> 
> Yes detailed feasability study is very important when kick starting a project 😉

Well, now I know! I succumbed to the fallacy of looking at ESP3D and thinking it was just like OctoPrint.

> Yes printing from WiFi is very bad idea ^_^, I only print from SD

I sort of want to try it just to show a web page slicing and then sending a print to a 3D printer, but I realize it may not be practical at all for long prints, due to generally flakiness of wifi.

> I have started the host code but need to redo it as not working as expected - also it will only work on ESP32 to be able to use wifi in same time without issue - forget esp8266 for such purpose IMHO

That makes sense. The ESP32 has two cores, if I recall correctly.

> That is really overkill for just USB stick support!!! I tested it before and I decided on ESP32 to use the CH376S instead and this lib: https://github.com/djuseeq/Ch376msc

Tell me about it. But I was working for LulzBot, which has some very odd rules about open source. The CH376S (which I recall looking at) appears to be a Chinese micro-controller running some closed-source firmware. This was nixed for not being open enough and the second choice was the MAX3421E. I spent the better part of a year trying to cram the USB host library into Marlin. The 8-bit boards can't really manage it, but the 32-bit Archim performs well with it.

marciot, 2020-07-30:
> FYI you should read :
> #239
> and [luc-github/ESP3D-WEBUI#94 (comment)](https://github.com/luc-github/ESP3D-WEBUI/issues/94#issuecomment-657225602)
> 
> To get latest status

Oh... a lot more is done than I thought. When I connect, I don't get the dashboard:

[image]

Maybe this has to do with the fact I am working on dev boards without an actual 3D printer connected?

luc-github, 2020-07-30:
you need to define target printer  Fw in settings  to get corresponding UI, if not  UI is minimal

marciot, 2020-07-30:
> you need to define target printer Fw in settings to get corresponding UI, if not UI is minimal

Ah. That was it. Thanks.

luc-github, 2020-07-30:
also UI is also dependding of what printer output in M408 or M105
still a lot to do to  be beta state....

luc-github, 2020-07-31:
May I know what MCU your printer board is using ?

marciot, 2020-07-31:
Our board is using Atmel sam3x8e

So, I've decided to try porting the serial host functionality from Cura into the ESP32, since I know that code works and handles things like Marlin serial errors and pause and resume pretty well. If I can get that to work in a standalone Arduino sketch, then I have a basic proof of concept.

I plan to use the basic ESP32 Web Server example as a starting point, as it interfaces with the SD card and already has a way to upload files to upload files to the SD card. I'll then see if I can start a thread on the second core to send the GCODE to the printer.

luc-github, 2020-08-01:
wow big task, I look forward to see the result
good luck :)

luc-github, 2020-08-10:
@marciot any update ?

marciot, 2020-08-10:
@luc-github:

I'm still experimenting and have been learning a lot about the ESP32 and it's limitations :)

I was able to adapt some earlier code I wrote for the Toshiba FlashAir to make a very basic UI for uploading files to the SD card.

Then I got a bit side tracked with an idea of using a Java service worker to fetch a single zip file from the ESP32 and unzip it into the browser's cache, so that the ESP32 had to do virtually no work at all to serve up the UI and it would only happen once, the app later persisting on the user's browser. I had a whole proof-of-concept working on my PC but when I put it on the ESP32, I learned that service workers only work over HTTPS -- so a lot of wasted effort there, but here is my experiment if you want to play with it:

https://github.com/marciot/service-worker-cache-from-zip

Anyhow, as for actually streaming GCODE to the printer, I got a simple thread running on a core to flash a LED. Not much so far, but it's a start!

I've also been pondering security a bit. I really dislike the idea of having an IoT device that is capable of burning down your house being unsecured and open. Would be nice if there was a way to encrypt the GCODE upload so only authorized users can do it, but learning how to do cryptography on the ESP32 is yet another thing to learn about.

luc-github, 2020-08-10:
>Would be nice if there was a way to encrypt the GCODE upload so only authorized users can do it, 

ESP3D support authentication, is it not what you need  instead of encryption ?

marciot, 2020-08-10:
> ESP3D support authentication, is it not what you need instead of encryption ?

I don't really know the ESP3D API that well. Are the API calls authenticated, or just the web UI? If it's just the web UI, then it's fairly easy to get around it.

> > HTTPS
> 
> I plan to add support to it after 3.0

I know the ESP3D can do HTTPS. But it is unclear [whether it can be done well](https://neosmart.net/blog/2017/lets-stop-punishing-iot-devices-that-embrace-https-shall-we). I don't know whether the concerns in that article have been addressed.

But anyhow, if you do get ESP3D to serve itself via HTTPS, making ESP3D into a persistent web app would be amazing. It would make it super fast. I don't know whether you've experimented with PWA and Service Workers -- it's kind of daunting to learn, but super powerful stuff.

luc-github, 2020-08-10:
Authentication need to be enabled in FW itself: https://github.com/luc-github/ESP3D/blob/2.1/esp3d/config.h#L89
There are 2 users : admin and user, protection is for all commands and pages

here example of check: https://github.com/luc-github/ESP3D/blob/2.1/esp3d/syncwebserver.cpp#L1215

but the upload by curl with authentication is supported only in 3.0,  in 2.1 it will be rejected

But it is not full proof, and this is out of my target to be honest

Arduino ESP32 core webserver does not support HTTPS currently( IDF support it but need porting to arduino way) that is why I opened : https://github.com/luc-github/ESP3D/issues/492

marciot, 2020-08-10:
@luc-github: I came across this the other day, but have not experimented with it:

https://github.com/fhessel/esp32_https_server

luc-github, 2020-08-10:
Yes this is the one I mentioned in https://github.com/luc-github/ESP3D/issues/492 and I already asked  a feature request ^_^ : https://github.com/fhessel/esp32_https_server/issues/95

marciot, 2020-08-10:
> @luc-github: I came across this the other day, but have not experimented with it:
> 
> https://github.com/fhessel/esp32_https_server

@luc-github: Ah. I see. You've been at this far longer than I have -- I've got a lot to learn and I'll probably end up walking the same path as you have :)

luc-github, 2020-08-10:
@marciot Na I just use ESP for a while now ;)

marciot, 2020-08-20:
I just wanted to let you know that I've been hard at work on an ESP32 print host. The code has been written and manual testing with the code via the console suggests it does what it is supposed to do. I now plan to hook it up to Marlin and go for full scale testing.
I have designed it as an Arduino library, so it should be very easy for you to integrate it into ESP3D if you so desire. It supports Marlin serial error correction/resend, pausing/resume/cancel, Marlin/RepRapFirmware flavors, ADVANCED_OK, action commands and the ability to run as a thread on any ESP32 core. The code is based on earlier work I did for USB hosts in Cura, so it should be fairly robust.
-- Marcio

luc-github, 2020-08-20:
nice - what is the library name ?

luc-github, 2020-08-30:
@marciot do you want me to test your print host ?

marciot, 2020-08-30:
@luc-github: Sorry about that, I thought I had replied to your earlier message. I have not been authorized to release the code to the public yet, but if you want to send me a private e-mail (see my GitHub profile) maybe I can get you a pre-release copy for testing.

luc-github, 2020-08-30:
NP - I understand, take your time there is no hurry - I will wait you release the code as I also would like to see how to add smoothieware and repetier support as you mentioned it is mainly for Marlin

luc-github, 2020-09-04:
I think issue can be closed now

closed 2020-09-04 by luc-github

3dlabsio, 2020-09-10:
> Yes detailed feasability study is very important when kick starting a project 😉
>
>
> Yes printing from WiFi is very bad idea ^_^, I only print from SD
>
> I have started the host code but need to redo it [...]

Will ESP3D have support for the CH376S using the library you mentioned in 3.0?

luc-github, 2020-09-10:
Not yet. But planned with this identified library

3dlabsio, 2020-09-10:
I'm looking to offer a bounty on that functionality in the current version (2.1). Similar to the Panucatt version with SD card. Let me know if you're interested or have a working test version.

luc-github, 2020-09-10:
but what would be the purpose in 2.1 to have usb disk support? printer cannot access to it

3dlabsio, 2020-09-10:
> but what would be the purpose in 2.1 to have usb disk support? printer cannot access to it

I can explain in more detail - is there an email I can reach out to you and take the discussion offline there?

luc-github, 2020-09-10:
you can use discord https://discord.gg/h2pkaf 

or email: luc at tech-hunters (dot) com