luc-github/ESP3D #488: [BUG] Unable to compile 3.0 branch due to missing "SPIFFS.h"

marciot, 2020-07-29:
**Describe the bug**
I was able to build the default branch, but unable to compile 3.0 branch due to missing "SPIFFS.h"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Hoping it will build.

**Screenshots**
```
Processing esp8266 (platform: espressif8266; board: esp12e; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 2.6.1 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20703.0 (2.7.3)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting esp3d.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <Ticker> 1.0
|-- <DHT sensor library for ESPx> 1.0.6
|-- <Wire> 1.0
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <TFT_eSPI> 1.4.11
|   |-- <SPI> 1.0
|-- <lv_arduino> 2.0.3
|-- <ESP8266SdFat> 1.1.0
|   |-- <SPI> 1.0
|-- <LittleFS(esp8266)> 0.1.0
|-- <SD(esp8266)> 2.0.0
|   |-- <SDFS> 0.1.0
|   |   |-- <SPI> 1.0
|   |   |-- <ESP8266SdFat> 1.1.0
|   |   |   |-- <SPI> 1.0
|-- <ESP8266SSDP> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266-Arduino-Lua> 0.0.30
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <WebSockets> 2.1.4
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Hash> 1.0
Building in release mode
Warning! '-Wl,-T' option for specifying linker scripts is deprecated. Please use 'board_build.ldscript' option in your 'platformio.ini' file.
Compiling .pioenvs/esp8266/src/esp3d.ino.cpp.o
Compiling .pioenvs/esp8266/src/src/modules/display/lv_flash_drv.cpp.o
Compiling .pioenvs/esp8266/src/src/modules/filesystem/esp_globalFS.cpp.o
Compiling .pioenvs/esp8266/src/src/modules/filesystem/esp_sd.cpp.o
Compiling .pioenvs/esp8266/src/src/modules/filesystem/flash/fat_esp32_filesystem.cpp.o
In file included from esp3d/src/modules/display/../../include/esp3d_config.h:27:0,
                 from esp3d/src/modules/display/lv_flash_drv.cpp:20:
esp3d/src/modules/display/../../include/../include/sanity_esp3d.h:82:2: warning: #warning Filesystem time is not available in SPIFFS ESP8266 yet [-Wcpp]
 #warning Filesystem time is not available in SPIFFS ESP8266 yet
  ^
In file included from esp3d/src/modules/filesystem/../../include/esp3d_config.h:27:0,
                 from esp3d/src/modules/filesystem/esp_globalFS.cpp:21:
esp3d/src/modules/filesystem/../../include/../include/sanity_esp3d.h:82:2: warning: #warning Filesystem time is not available in SPIFFS ESP8266 yet [-Wcpp]
 #warning Filesystem time is not available in SPIFFS ESP8266 yet
  ^
In file included from esp3d/src/modules/filesystem/../../include/esp3d_config.h:27:0,
                 from esp3d/src/modules/filesystem/esp_sd.cpp:21:
esp3d/src/modules/filesystem/../../include/../include/sanity_esp3d.h:82:2: warning: #warning Filesystem time is not available in SPIFFS ESP8266 yet [-Wcpp]
 #warning Filesystem time is not available in SPIFFS ESP8266 yet
  ^
In file included from esp3d/src/modules/filesystem/flash/../../../include/esp3d_config.h:27:0,
                 from esp3d/src/modules/filesystem/flash/fat_esp32_filesystem.cpp:20:
esp3d/src/modules/filesystem/flash/../../../include/../include/sanity_esp3d.h:82:2: warning: #warning Filesystem time is not available in SPIFFS ESP8266 yet [-Wcpp]
 #warning Filesystem time is not available in SPIFFS ESP8266 yet
  ^
esp3d/src/modules/display/lv_flash_drv.cpp:27:20: fatal error: SPIFFS.h: No such file or directory

****************************************************************
* Looking for SPIFFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPIFFS.h"
* Web  > https://platformio.org/lib/search?query=header:SPIFFS.h
*
****************************************************************

 #include "SPIFFS.h"
                    ^
compilation terminated.
*** [.pioenvs/esp8266/src/src/modules/display/lv_flash_drv.cpp.o] Error 1
===================================================================== [FAILED] Took 10.18 seconds =====================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       SUCCESS   00:02:57.617
esp8266        FAILED    00:00:10.183
================================================================ 1 failed, 1 succeeded in 00:03:07.800 ================================================================
```

**ESP3D Firmware:**
- ESP core version: 3.0

**Target Firmware:**
 - Name: Marlin
 - Version 2.0

**Board used (please complete the following information):**
 - MCU: ESP8622
 - Name: ESP-01S 
 - Flash size: 1MB

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

marciot, 2020-07-29:
I was able to get it to compile by changing:

`#define FILESYSTEM_FEATURE ESP_SPIFFS_FILESYSTEM`

To:

`#define FILESYSTEM_FEATURE ESP_LITTLEFS_FILESYSTEM`

I also disabled `#define DISPLAY_DEVICE TFT_SPI_ILI9488_480X320` (not sure why that is default) and disabled `#define NOTIFICATION_FEATURE` to make it fit on the ESP-01S.

So, anyhow, it does work with config file changes. Not sure whether it is considered to be a bug that it doesn't compile without changes.

luc-github, 2020-07-29:
what os are you using to compile?

marciot, 2020-07-29:
Ubuntu under WSL 1 on Windows 10

luc-github, 2020-07-30:
I need to check as it is fine on my window and travis also says compilation is passing

luc-github, 2020-07-30:
Ok my mistake , seems I only used ESP32 for display and there is missing define for esp8266

luc-github, 2020-07-30:
anyway advanced display looks not fit the esp8266 I have `section `.rodata' will not fit in region `dram0_0_seg'` error
so yes disabling advanced display on esp8266 it may be a must - but the initial target is esp32 so it should be fine

Closed as fixed - thank you for pointing this out

closed 2020-07-30 by luc-github

marciot, 2020-07-30:
No problem! Glad I could help!

commit fa2b2916 (2020-07-30): Fix  compilation error due to "SPIFFS.h" in esp8266

thanks @marciot  for pointing this
Fixes #488