luc-github/ESP3D #489: [BUG] Unable to connect after accidentally uploading "index.html.gz" for 2.1 to ESP3D 3.0

marciot, 2020-07-29:
**Describe the bug**

When the board first boots up when running ESP3D 3.0, it requests that you upload "index.html.gz". However, I accidentally uploaded the version for "index.html.gz" for 2.1

Once I did this, the web page said it was unable to connect. Reflashing ESP3D 3.0 did not clear the file system, so I had to resort to this trick to wipe the Flash clean by uploading a 1M blank file:

http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/#:~:text=If%20you%20want%20to%20erase,as%20clearing%20the%20entire%20flash!

It seems like the "Unable to connect" dialog box should have an option to re-upload the "index.html.gz", otherwise there is not an easy way to recover from the situation.

**To Reproduce**
Steps to reproduce the behavior:
1. Install ESP3D 3.0
2. Upload "index.html.gz" for 2.1
3. Try to connect to ESP3D

**Expected behavior**
There should be a way to recover from this situation without having to use special tools to wipe the Flash clean.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:**
- ESP core version: 3.0
- Version: 3.0 branch
- Web UI Version: 2.1  

**Target Firmware:**
 - Name: Marlin
 - Version 2.0

**Board used (please complete the following information):**
 - MCU: ESP8622 
 - Name:ESP-01S
 - Flash size: 1M

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

marciot, 2020-07-29:
Okay. So I got ESP3D 3.0 to run, the UI looks nice, but seems like there is a lot of work yet to be done.

When I connect via web sockets I get "activeID:0" and "activeID:1". Not sure what that is, but at least it seems to work.

luc-github, 2020-07-30:
use the maintenance page : `http://<yourIP>?forcefallback=yes`

[image]

you do not use the proper websocket you are currently using the websocket for webui communication not for bridge so what you see is communication between webUI and esp

luc-github, 2020-07-30:
I close issue as same as https://github.com/luc-github/ESP3D-WEBUI/issues/107

closed 2020-07-30 by luc-github