luc-github/ESP3D #491: [FEATURE REQUEST]Add WebDAV feature

luc-github, 2020-07-30:
Use version from @d-a-v esp8266 guru : https://github.com/d-a-v/ESPWebDAV 
could be a good alternative to ftp server and lighter in term of code

should work also with esp32 but may need some tuning

No version set yet as need to finish the current todo list for 3.0

vashilov, 2020-09-16:
https://github.com/ardyesp/ESPWebDAV
Sorry about my English.
This is an interesting scheme for connecting sd card to ESP and 3D printer at the same time.

It may work without webdav, but you need to test it.

luc-github, 2020-09-16:
I know this one, it is used by : https://github.com/luc-github/ESP3D/issues/454
and the git you refere is the base of https://github.com/d-a-v/ESPWebDAV

luc-github, 2021-01-19:
As it is only supported by esp8266 I have started the esp32 porting :
- the webserver hook is done https://github.com/luc-github/arduino-esp32/commit/b5233795d7e61a2fc724397bb61db7065921bb41
- webdav part is also done https://github.com/luc-github/ESPWebDAV/tree/ESP32

but I have found some issue in original code :  https://github.com/d-a-v/ESPWebDAV/issues

also it seems for some reason the webserver does not see windows mount point queries when same queries are seens when sent from webbrowser and hook is working

So I may consider to not use the webhook version but a dedicated server instead like for ftp 
Also the lib is defined to work with specific FS and SPIFFS is not really a good candidat so far
I would like to use the Global FS of ESP3D instead to allow to show FS and SD under the root
But before doing this need the issues I have raised to be fixed so TBC

luc-github, 2021-01-21:
Ok some update:
I have fixed the issues in original code and now directories / parsing act properly but SPIFFS due to virtual FS, so WebDav is now working on ESP32
Hook web server POC  is also working on esp32, but the hook itself is not the problem, some additional feature like HTTP_HEAD support, connection keep-alive, and may be another one, are not supported - I did my best https://github.com/luc-github/arduino-esp32/tree/webserver-hook, but the mount point is still not successful and this is beyond my knowledge 😭 

So I may consider to use a standlone webdav solution using another server than webserver and another port than webserver one, this would allow to not stick to any webserver solution - TBC

luc-github, 2021-01-23:
Now implementing WebDav code as standalone (not webserver hooked) it will serve FS and SD (Direct / Shared)
The WebDav won't be implemented for Serial SD as too slow to get answer and cannot upload really fast, on MKS protocol the upload is faster but unfortunalty it cannot manage files commands like copy / rename / delete / directories commands / time / size so it is out of scope also.
It could be added in future for MKS TFT which would allows more flexibility - for Marlin and related I do not know

So target is currently limited but I hope it can extend to other configs when prerequisites are met

pipi61, 2021-01-23:
Hello!
Recommendation: The 128-byte buffer size is too small for transmission. I tried to change LITTLEFS to SD_MMC and then 140kByte/s transfer rate. If the buffer size is 4k, the speed is up to 500kByte/s
Buffersize changed in ESPWebDAVCore::handleGet(...) and ESPWebDAVCore::handlePut(...) functions
---
  there are no issues tabs at https://github.com/luc-github/ESPWebDAV page.
---
I've tried several other webdav examples on esp32, but I can't port from another proc, or there's a bad job in explorer or TC.
If the webdav plugin is installed in TotalCommander, your example (Simple_esp32.ino) work fine! (and works well in browser and explorer) 

very good product
Thank You,

luc-github, 2021-01-23:
Hi thank you
there is no issue tab as it was for PR to  https://github.com/d-a-v/ESPWebDAV - the  repo will be deleted now PR is merged .
Can you open your issue/ suggestion here : https://github.com/d-a-v/ESPWebDAV/issues

On ESP8266 I am afraid cannot use 4K, but on ESP32 should not be a problem

luc-github, 2021-01-24:
OK now webDav is part of ESP3D for esp8266 and esp32  - may be need some tuning but it is working - now need some test as only tested on esp32 littlfs - need to check all FS on esp32 and esp8266 which is quite number 
as SPIFFS is deprecated and does not support Directory  - the support is only on root not on subdirectories, littlfs is available already as external lib in 1.0.5 and in 2.0 it will be fully integrated so I think not put all effort in SPIFFS support is ok

@pipi61 I have increased buffer for esp32 per your recommendations but currently only to 1024 instead of 4096, (4096 is TBC)   - I did not do any bench but indeed I feel response is faster

luc-github, 2021-01-24:
OK implemented and tested on esp32 littlefs and esp8266 littlefs  -  need more test for sd and larger file so may be need more tuning but feature is there so I close issue

closed 2021-01-24 by luc-github

luc-github, 2021-01-24:
Additional feedback:
Tested with ESP32 16MB fat + 8MB PSRAM + SD 4GB and global FS
Upload of 1MB FIle is ok so far and listing is working on GFS FAT and GFS SD

pipi61, 2021-01-24:
Feedback: I try using SD_MMC library, 18,5Mbyte filesize, on esp32 your webdav library.
i don't know the optimal buffersize 1. SD sectorsize*x , or  wifi transferblocksize*x. I try change 128 to 1*1440, 2*1440,3*1440, 4*1440 buffersize. 1*,2*,3* work good, But change to 4*1440 buffersize, the esp32 restart if begin file transfer.
Thank Your feedback.

luc-github, 2021-01-24:
@pipi61 thank you for your feedback, so based on your experiment 4K is the limit (3*1440 )

Do you see siginicative performance difference between a buffer of 1024 and a buffer of 4096 ?

pipi61, 2021-01-24:
[image]

luc-github, 2021-01-24:
sdmmc is faster than sd.
wow with 4k vs 1k the speed is x2

pipi61, 2021-01-24:
and speed up, if tune sd_mmc library
[image]
[https://github.com/espressif/arduino-esp32/issues/1611](url)