luc-github/ESP3D #492: [FEATURE REQUEST]Add support for alternate webserver library

luc-github, 2020-08-02:
There is nice library for esp32 that seems more close in feature from esp8266 than current embedded esp32
https://github.com/fhessel/esp32_https_server

Seems even support async way and several clients - need to benchmark it but it may be the solution for the slow upload and file download in http way vs embedded esp32 webserver

Last but not least it supports HTTPS which is in future must be implemented in ESP3D 

for transition period  it even provide a wrapper : https://github.com/fhessel/esp32_https_server_compat, and add experimental support of esp32 websocket

And it is actively supported

so all is benefit ^_^

Need to do some bechmark but very promising

ETE-Design, 2020-11-07:
@luc-github  Have you testede this yet? Would be great to have faster file transfer...

luc-github, 2020-11-07:
no I did not 
the goal of this is https - not sure the file transfer will be faster with this server

luc-github, 2020-11-07:
the plan of this will be after 3.0 so may be 3.1

luc-github, 2021-02-04:
\O/
  |  AsyncWebserver is back in game https://github.com/me-no-dev/ESPAsyncWebServer
 /\

luc-github, 2021-02-19:
Additionnal update: SecureWebServer is coming in 2.0 esp32 core  😸 
  https://github.com/espressif/arduino-esp32/issues/3902#issuecomment-781258806

luc-github, 2022-03-20:
additional potential library to bench for ESP32 https://github.com/khoih-prog/WiFiWebServer

luc-github, 2023-03-22:
[ESP3D-TFT](https://github.com/luc-github/ESP3D-TFT) use light httpd and so far so good - did not tried the https but it is well supported, I have coded the multipart POST support so moving to idf when esp8266 is end of life will definitly the way to go

luc-github, 2023-10-05:
This was open for a long time and no solution so far so let's close it

closed 2023-10-05 by luc-github