luc-github/ESP3D #493: [BUG] ESP-02(s)  not starting

Exellent1988, 2020-08-10:
**Describe the bug**
Cant get the 3.0 alpha version to run on my ESP-01(s) ( got a few)
But i can get it to start on en ESP8266MOD (NodeMcu Dev Board).
The only thing i get is M117 Wifi Off
cant get it to turn the Wifi on,by ESP Commands or anything else.

Do you have any Ideas?

**To Reproduce**
**ESP3D Firmware: 3.0
- Wifi mode:[ STA]  
- Any change done in FW  None

**Target Firmware:**
 - Name: [e.g. Marlin]
 - Version [e.g. 2.0]

**Board used (please complete the following information):**
 - MCU: [ESP826] 
 - Name:[e.g. ESP01] 
 - Flash size: [e.g. 1M]

luc-github, 2020-08-10:
what flash size for file system do you set ?

Exellent1988, 2020-08-10:
So as i understand it correctly, the android-core version doesnt mather if you use plattform.io ( which i do).
So but after try to figure out what flash size i used, it seems that it was defined to 4m ^^ so just for refernce:
Thats the working env for an ESP-01(s) :)

```
[env:ESP-01]
platform = espressif8266
board = esp01_1m
upload_protocol = esptool
framework = arduino
monitor_speed = 115200
; set frequency to 160MHz
board_build.f_cpu = 160000000L
; set frequency to 40MHz
board_build.f_flash = 40000000L
board_build.flash_mode = dio
upload_resetmethod = ck
build_flags = 
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore = 
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD
    SdFat
```

luc-github, 2020-08-10:
android-core? 

You removed the 4MB flash 2MB SPIFFS, but you do not specified the FS size

Aurel-14, 2020-08-11:
hy,

You can try this code below.
Just change few lignes in esp8266 env

`
[env:esp8266]
platform = espressif8266
board = esp01_1m
framework = arduino
monitor_speed = 115200
; set frequency to 160MHz
board_build.f_cpu = 160000000L
; set frequency to 40MHz
board_build.f_flash = 40000000L
board_build.flash_mode = dout
upload_resetmethod = nodemcu
board_build.ldscript = eagle.flash.1m256.ld
build_flags =
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore = 
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD
`

luc-github, 2020-08-11:
@Exellent1988 does your issue is solved ?

Exellent1988, 2020-08-11:
the version which woreked for me was now the following config:

```
[env:ESP-01]
platform = espressif8266
board = esp01_1m
upload_protocol = esptool
framework = arduino
monitor_speed = 115200
; set frequency to 160MHz
board_build.f_cpu = 160000000L
; set frequency to 40MHz
board_build.f_flash = 40000000L
board_build.flash_mode = dio
upload_resetmethod = ck
board_build.ldscript = eagle.flash.1m512.ld
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore = 
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD
    SdFat
```

luc-github, 2020-08-11:
Really? what size is your FW ?
I see you use 512KB for FS which means your FW should be less than 500K to fit the flash 
but on my side, the compilation give me more than 500KB for the FW, so you should have some overlap in flash that could give erratic behavior
a FS around 192KB/256KB  like above should be better

luc-github, 2020-08-18:
if no update I close issue

closed 2020-08-18 by luc-github