luc-github/ESP3D #495: Won't compile in VSCode>

Drzet, 2020-08-28:
Errors in ...
.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:271:12: error: 'isnan' has not been declared in 'std'
.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:270:12: error: 'isinf' has not been declared in 'std'

Visual Studio Code and platformIO latest versions, all libraries updated. Windows 10 x64
ESP3D version 2.1.1.1.

luc-github, 2020-08-28:
for me everything is Ok 
[image]

also the error described looks like package issue 

what dio you mean by `all libraries updated` ?

Drzet, 2020-08-28:
It means I have clicked on "update project libraries" in platformIO

luc-github, 2020-08-28:
Ok this feature has no impact in ESP3D project as libraries are no referenced but in libraries directory

so your problem is likely an installation problem in platformIO of esp8266 package - as Arduino.h is from esp8266 package itself

Drzet, 2020-08-28:
No idea. I managed to compile the firmware in ArduinoIDE.

luc-github, 2020-08-28:
Ok great - I close issue then as it is not duplicate

closed 2020-08-28 by luc-github

sl1pkn07, 2020-09-04:
put this in the platformio.ini env section

~~~
platform_packages      = toolchain-xtensa@2.40802.200502
~~~

seems `2.100100.200706` version have problems

sl1pkn07, 2020-09-04:
seems add `-std=gnu++11` in the build flags instead set the `platform_packages` option also fix the problem

luc-github, 2020-09-11:
@sl1pkn07 indeed I got same issue when I was requested to update extension and PIO 
Thank you for the tips - it solved it
and now with latest update I also cannot select the target environement anymore from GU.
I it only build `default_envs` or all env...

So need to manualy change `default_envs = esp32dev` to `default_envs = esp8266` to build only one

luc-github, 2020-09-11:
In 3.0 I have added custom tasks to build esp8266dev and esp32 separatly
Not sure it is the right way to do but it works....

```
{
			"type": "shell",
			"command": "platformio",
			"args": [
				"run",
				"--environment"
				"esp32dev"
			],
			"problemMatcher": [
				"$platformio"
			],
			"group": "build",
			"label": "PlatformIO: build (esp32dev)"
		},
		{
			"type": "shell",
			"command": "platformio",
			"args": [
				"run",
				"--environment"
				"esp8266dev"
			],
			"problemMatcher": [
				"$platformio"
			],
			"group": "build",
			"label": "PlatformIO: build (esp8266dev)"
		},
```

sl1pkn07, 2020-09-11:
[image]

without touch nothing in Code

[image]

luc-github, 2020-09-11:
I usually go to terminal tasks
[image]

But once I go to platformIO button  like you showed - the menu is updated...
[image]

I am not super familiar with platformIO, I never used this platformIO button on left side before 😅  - thank you so much for teaching me this @sl1pkn07  - again you solved a troublesome behavior  - thank you 😸 

I will remove the custom tasks then