luc-github/ESP3D #496: WeMOS_Mega_2560_ESP2866_WIFI

jojoxyz, 2020-08-30:
My problem is communication. Printer makes 2 (Lines) commands and stops.
I've tried all possible configurations and always the same.
Does somebody has any idea?

WeMOS_Mega_2560_ESP2866_WIFI   ,   Ramps 1.4  ,  Repetier v 0.92

luc-github, 2020-08-30:
sorry can you give more details of your issue ?
what do you do ? and what happen ?

It is not clear

jojoxyz, 2020-08-31:
Whenever I execute G28, everything goes ok, but then no command works anymore. Or if I drive manual it works with 1 or 2 movements and then no longer reacts. When I start printing, the printer starts to heat but does not start to press.

luc-github, 2020-08-31:
Ok weird - more questions.
* What is your Printer FW ? 
* What is your ESP8266 board ?
* Wow did you wired esp to ramps ?

when you start print ESP3D does not do anything after so if printer heat but do not actually anything the problem is on printer side (FW or Slicer)
Fromt another point of view when heating most of printer FW reject any commands until temperature is reached 

* Please copy and past the `[ESP420]plain` output from terminal command

 * when commands stop working :
1 - does `[ESP420]plain` command still give out put ?
2 - what is the terminal output when sending `M115` ? 
3 - does your printer still  works from your printer menu ?

jojoxyz, 2020-08-31:
Controller  MEGA+WiFi R3 ATmega2560+ESP8266, flash 32MB, USB-TTL CH340G, Micro-USB
https://robotdyn.com/mega-wifi-r3-atmega2560-esp8266-flash-32mb-usb-ttl-ch340g-micro-usb.html

FW     Repetier v 0.92

luc-github, 2020-08-31:
what change did you do in repetier to make comunication working  between mega and esp?

does this happen using telnet client like putty and 8888 port ?

jojoxyz, 2020-08-31:
but nothing works.
I made small progress with memory change and time setting in repetier host
actually everything is configured correctly but it just doesn't work.

luc-github, 2020-08-31:
>actually everything is configured correctly .

then it should work

> but it just doesn't work

So it is not configured correctly

>I made small progress with memory change and time setting in repetier host

so you do not use webUI ? only repetier host and tcp 8888 port ?

jojoxyz, 2020-08-31:
I tried standard 8888 and others but nothing. In mega bluetooth serial 3
web ui makes the same problems.

luc-github, 2020-08-31:
what is baudrate on bt seriat 3

jojoxyz, 2020-08-31:
115200
I also tried 250000
no effect

luc-github, 2020-08-31:
by default in FW it is 115200 so esp need to be also 115200 - I guess you be sure both match ?

it may be a bug of repetier V0.92 - now repetier is 1.0.4 final did you tried it ?

jojoxyz, 2020-08-31:
both values are the same. i am trying fw repetier 1.0.4 but with the same result

luc-github, 2020-08-31:
well I have not clue what going on - I do not know this board 

Is the usb disconnected ? I see depending of switch the USB is also sused by ESP

jojoxyz, 2020-08-31:
the problem is: if i separate ramps from mega it works. shows temperatour, loads fw config.

luc-github, 2020-08-31:
so it is switch problem ?

jojoxyz, 2020-08-31:
ON | ON | ON | ON | OFF | OFF | OFF

all functions are ON

jojoxyz, 2020-08-31:
problem is internal configuration in fw (pins), but no idea how icg can solve the problem

luc-github, 2020-08-31:
what about on on off off off off off  off?

jojoxyz, 2020-08-31:
no other configuration does not work. mega must have access to pins and wifi at the same time. that only works in my configuration. I tried it with blink led / server

jojoxyz, 2020-08-31:
problem solved. tx3 pin 13, 14 connected at ramps on y min, y max.

luc-github, 2020-08-31:
great !
happy you found out

closed 2020-09-01 by jojoxyz