luc-github/ESP3D #498: [FEATURE REQUEST]<Add support for TCP connection or Rest API>

GCarneiroA, 2020-09-07:
Add TCP Server or API Rest for use with third party applications, own software, without the need to use the ESP3D-WEBUI web interface.

GCarneiroA, 2020-09-07:
There is not much to say here, I believe the context is self-explanatory, I need to connect my desktop application via TCP to ESP3D and it is not possible without a TCP server.

luc-github, 2020-09-07:
TCP server is already implemented :
Features
* Serial/Wifi bridge using configurable port 8888

and you can use direct commands: https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands

it is enabled by default in 2.1 
https://github.com/luc-github/ESP3D/blob/2.1/esp3d/config.h#L54-L55

only file upload must use http by POST but you can use curl like API

Rest API is not planed

luc-github, 2020-09-09:
I am not related to BTT so I cannot say what they enabled or not and they may use an old version of FW

yes it has be implemented to work with repetier host and pronter face, but for streaming you need to adjust you printer cache / sender cache and having a good network and QoS for such streaming 
TCP may have some latency and if your parameters are not compensating you may have random pause and so some blob will happen on your print

Even some people do and are happy with (https://github.com/luc-github/ESP3D/issues/387#issuecomment-557202401), I do not recommend printing over tcp, printing from SD is the safest way IMHO

luc-github, 2020-09-10:
I guess issue can be closed as answered

closed 2020-09-10 by luc-github