luc-github/ESP3D #500: [FEATURE REQUEST]More github Automation

luc-github, 2020-09-07:
I plan to implement more github automation in workflow and issue

I am not sure what exactly but the feature looks powerful and I am not happy with the current tools
- stale : so far so good but github-action may do better, I am thinking to complete with autoclose https://github.com/marketplace/actions/auto-close-issues

- chat : I plan to move from gitter to another chat tool (https://github.com/luc-github/ESP3D/issues/499) which will also need github-action

- TBD 

Feedback / ideas are welcome to such improvement

nullstalgia, 2020-09-22:
Autoclose closed https://github.com/luc-github/ESP3D/issues/427 , so new images can't be added to the wiki.

luc-github, 2020-09-22:
good point -  unlocked, thank you ^_°

luc-github, 2020-10-12:
Auto close has been implemented
Wiki is now also part of repo sync wiki directory when doing PR

I think this ticket can be closed now

closed 2020-10-12 by luc-github