luc-github/ESP3D #501: [Question]What is it camera_device command for in ESP401?

royfocker19, 2020-09-09:
Hi Luc, I’m not able to make the camera work on an esp32cam from diymore.
Searching for camera related things on the code I found this:

#ifdef CAMERA_DEVICE
                case ESP_CAMERA_PORT:
                    //esp3d_camera.begin();
                    break;
#endif //CAMERA_DEVICE

Is it ok to have that line commented? It’s going to be used to enable the server on demand? Now the server is always on?
Thanks in advance

luc-github, 2020-09-09:
the ESP401 is the command to change setting in EEPROM, for some settings I try to avoid to ask user to restart board when possible to apply them as hot change

in that case the camera port change - but not all features like hot change - when I tested the camera did not worked well if board was not reset when changing the http port - so I commented it and kept as not to check again when esp32 core has update to see if it was fixed when I will work on camera again

So to sum up when you change EEPROM setting for camera you must restart board to apply the change, so yes it is ok for the moment

royfocker19, 2020-09-10:
Ok, so in that case maybe it has something to do with pins configuration or something else. I was able to make it work with the camera web server sketch and the ledc patch, but it doesn’t work here.

Press in Dire Situations http://www.nooooooooooooooo.com/

luc-github, 2020-09-10:
May be I have no clue sorry - I close question as answered  - fell free to open new issue if you need

closed 2020-09-10 by luc-github