luc-github/ESP3D #503: [BUG]Buttons don't work from Android website

Liger0, 2020-09-11:
**Describe the bug**
When trying to connect to the web UI interface from android, the buttons to upload don't work.
In my case, I don't have wifi on my PCs as I use ethernet, so I tried to upload the firmware via the Phone.
But the upload buttons don't work there.
So I tried USB tethering, but the redirection fails on any browser, and it only works clicking on the "Click here if you aren't connec ting" message, but that page only shows up for a fraction of seconds so it's pretty hard to setup.
Also it isn't very clear on which of the 2 options you should click to upload, unless you know so beforehand.

**Expected behavior**
Android Web UI working

**ESP3D Firmware:**
- ESP core version: 2.7.4
- Version: 2.1.1.b1
- Web UI Version: 2.1b60
- Wifi mode: Router
- WebUI version: Already asked?

**Board used (please complete the following information):**
 - MCU: esp8266
 - Name: Wemos D1 Mini pro
 - Flash size: 16MB

luc-github, 2020-09-11:
looks like android captive portal now block upload buttons

luc-github, 2020-09-19:
I close issue as duplicate of https://github.com/luc-github/ESP3D-WEBUI/issues/52

closed 2020-09-19 by luc-github