luc-github/ESP3D #504: Control Tronxy

Wuckel01, 2020-09-13:
Hello,
it is possible to control the Tronxy firmware V1.4.1RC24m4 with the TronxyX5SA board  https://de.aliexpress.com/i/32920995172.html with the original firmware and an ESP8266 module.
I have the oginal module which only works directly with wlan and that is not exactly super stable.
I've already tried several settings in esp3d without success.

I have esp3d running on another printer with marlin and it's great.

luc-github, 2020-09-13:
What M115 give as output?

Wuckel01, 2020-09-15:
Thank you for your prompt reply
I did not get any interface connection. 
I've tested everything from 9800-25000
 
I couldn't run M115

luc-github, 2020-09-16:
yes I know, I mean from serial

Wuckel01, 2020-09-16:
Hello

if I control USB with Pronterface and the com connection with 115200bps and M115 I get an answer.

 

Connecting...

Printer is now online.

>>> M115 

SENDING:M115

ok CBD make it.Date:Jul 22 2019 Time:17:18:49

 

I can also control everything.

 

greeting

luc-github, 2020-09-16:
well looks like M115 is not supported - do you know what kind of FW is it ?

Wuckel01, 2020-09-16:
no is original from the manufacturer a firmware on it that is really great but the wlan module software and app is junk

Wuckel01, 2020-09-16:
Hello

 

it must be based on

https://shop.cbd-3d.com/download/ but has been modified. this is the manufacturer side

https://tronxy.com/

Wuckel01, 2020-09-16:
Hello

 

What I think is great about the original firmware is that if the power fails, I can continue printing at the last stand and that the print head can also be ausiert, which I have never managed with marin because the firmware is so big that it does not fit in the memory for this module.

therefore I wanted to keep the original firmware

luc-github, 2020-09-16:
well it looks like an esp01 board so only serial connection - but I read here it is duet wifi https://de.aliexpress.com/item/32990868079.html

Wuckel01, 2020-09-16:
Hello

 

I downloaded the original firmware from the original esp01 and copied it to a new one that works. I was able to establish a connection to cloonen with 115200.

 

there is a possibility if there is a duet wifi to use esp3d.

luc-github, 2020-09-16:
If you can share the zip of the FW I may have a look, but as I do not have this tronxy board my in estication will be limited

Wuckel01, 2020-09-16:
Hello

yes send you the firmware to no thing

luc-github, 2020-09-17:
I do not see the file, does M408 command give any output by serial?

Wuckel01, 2020-09-19:
Hello

did not have time.

This morning I hung myself between the esp-01 original firmware tronxy and the paltinge and recorded the connection

 

I added sceen
 
if I switch on the wifi on the tronxy and switch it off it comes out with 115200bps
 
it looks like he would adjust the interface speed when activating it

 

regards

 

Sascha

Wuckel01, 2020-09-19:
Hello

 

when I send via USB Connect M408 that happens in the Sceenshot

luc-github, 2020-09-19:
when you use email to answer the files are not visible, please use github to attach files

Wuckel01, 2020-09-19:
Hello

I thought that would get well, I'll hang it on

[image]
[image]

luc-github, 2020-09-19:
are you sure your connection is correct? seems printer does not answer to M115 neither M408
and even no error if not supported

Wuckel01, 2020-09-19:
Hello

So with the pronterface program I can drive home without any problems and also check the extruders.
[image]

Wuckel01, 2020-09-19:
Hello

I'll make a screen when heating up
[image]

Wuckel01, 2020-09-19:
Hello

I entered the commands again that one M114 outputs a date, the other only says N: 0 as in the screen, so the printer reacts to it, only probably there are other M
commands.
[image]

Wuckel01, 2020-09-19:
I still have a gcode file
the configuration of the printer with M commands

Wuckel01, 2020-09-19:
only once such an idea can you install a serial bit rate of 2250000 in esp3d?

luc-github, 2020-09-19:
no this is far too high

Wuckel01, 2020-09-19:
why does he ask when I turn on the wifi on the software
AT+UART_CUR=2250000,8,1,0,0

luc-github, 2020-09-19:
AT+UART_CUR=2250000,8,1,0,0
does not come from esp3d

Wuckel01, 2020-09-19:
no that comes from the printer software on the esp8266-01 module when I switch on the wlan function in the firmware on the lcd

HaernCalistinson, 2020-09-20:
Just for sake of clarity, it's Chitu firmware. Closed source, can't do anything with it. IMHO biggest mistake Tronxy ever made. Configured using gcode commands via a txt file on the sd card.

luc-github, 2020-09-20:
The I guess there is nothing that I can do - 

according here https://www.thingiverse.com/groups/tronxy/forums/general/topic:45811 
you should may be give a try to Marlin : https://github.com/MarlinFirmware/Marlin/pull/15493

luc-github, 2020-09-20:
I close issue then

closed 2020-09-20 by luc-github

HaernCalistinson, 2020-09-20:
I will second your suggestion to move the printer to Marlin. While it requires a bit more tuning, it opens up the printer to all sorts of possibilities and solves a lot of the problems that the printer has. The X5SA users group on FB has some pretty easy guides.

Wuckel01, 2020-09-20:
Hello

I've already had the Marlin firmware on it.
This is not an option for me because the printer with the original firmware is better.
Since Marlin only brings some functions if you switch on functions but the firmware is so big that it no longer fits on the board,
E.g. Autoleveling 6 points, save duckstand after a power failure, stop lifting your head in home and continue printing the deadline at most space, etc. what the original firmware can do.

I have a different solution but I have to order hardware first esp3d host USB so I can plug the esp3d directly into it and externally with an additional 5 V supply. I already wrote a housing in 3d for 5v stepdown 12V Mosfet for LED light and control via esp32 with esp3d, only usb host shield is missing.