luc-github/ESP3D #505: [BUG]<ESP3D-2.1 AUTHENTICATION_FEATURE>

vashilov, 2020-09-17:
If I enable AUTHENTICATION_FEATURE, I can not connect to default ESP3D AP.

If I use default Firmware, config ESP to connect to my Wifi network, it fine.
When I update Firmware from .bin (enable AUTHENTICATION_FEATURE) on web interface, ESP do not connect to my wifi.

P.S. AUTHENTICATION_FEATURE it is login and password to web interface?

**Screenshots**
[image]
[image]

**ESP3D Firmware:**
SDK:  2.2.2-dev(38a443e)
Flash Size:  4.00 MB
FW version:  2.1.0.b39 ESP8266/8586
WebUI version:  2.1b65

vashilov, 2020-09-17:
Sorry if I did something wrong.
But maybe it will be useful to someone.

vashilov, 2020-09-17:
P.S. Discord https://discord.gg/h2pkaf not work.

luc-github, 2020-09-17:
Yes Authentication Feature has no impact on wifi connection itself

Thank you for pointing out - new link  never ending is now https://discord.gg/Z4ujTwE

vashilov, 2020-09-17:
But I dont connect to ESP on wifi and ESP not connect to my router.
It is only for me bug?
ESP3D dont work for me with AUTHENTICATION_FEATURE

vashilov, 2020-09-17:
I only uncomment AUTHENTICATION_FEATURE

luc-github, 2020-09-17:
as requested in template:
**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- Flash method: [Arduino Ide / PlatformIO]
- Any change done in FW  

**Target Firmware:**
 - Name: [e.g. Marlin]
 - Version [e.g. 2.0]

**Board used (please complete the following information):**
 - MCU: [e.g. ESP32] 
 - Name:[e.g. NodeMCU2s] 
 - Flash size: [e.g. 4M: 2M/2M]

vashilov, 2020-09-17:
ESP3D Firmware:

    ESP core version: 2.7.4 ???
    Version:2.1.0.b39
    Web UI Version:2.1b65
    Wifi mode:STA and CLIENT
    Flash method: Arduino Ide 1.8.9
    uncommet #define AUTHENTICATION_FEATURE

Target Firmware:

    Name: Marlin
    Version 2.0

Board used (please complete the following information):

    MCU: ESP8266EX
    Name:clone NodeMCU and clone Wemod D1 mini
    Flash size: 4M: 2M/2M

vashilov, 2020-09-17:
Please send me bin file for ESP8266. I upload it on my ESP3D from web.

luc-github, 2020-09-17:
I cannot reproduce your issue with same hardware
[image]

Also you use old Fw - please use latest one from git

vashilov, 2020-09-17:
I use FW from default branch 2.1.
Try now from master.

Thanks for answer.

luc-github, 2020-09-17:
yes default branch is 2.1 is correct - but your wrote `Version:2.1.0.b39` so you use old version

vashilov, 2020-09-17:
I download source from branch is 2.1.
[image]
[image]

What I do wrong?

vashilov, 2020-09-17:
You use ESP3D-2.1-devt ?
Now I use it.
AUTH enable and my phone dont connect to wifi
[image]

AUTH comment
[image]
[image]

vashilov, 2020-09-17:
maybe I need to fix something else?
For example, admin password and the like.

luc-github, 2020-09-17:
No sorry I do not send binary on demand, if I start to do, every one will ask for it, and I have no reason and no time to server every one.

how come after flashing you do not have index.html anymore ?
and man you said you flash with Flash size: 4M: 2M/2M: but your UI show you use 3M for flash...

type and show the output of :
[ESP420]plain
[ESP800]
[ESP410]plain

vashilov, 2020-09-17:
[ESP420]plain
Chip ID: 12554327
CPU Frequency: 160Mhz
Free memory: 36.51 KB
SDK: 2.2.2-dev(38a443e)
Flash Size: 4.00 MB
Available Size for update: 536.50 KB(Ok)
Available Size for SPIFFS: 2.81 MB
Baud rate: 115200
Sleep mode: Modem
Channel: 11
Phy Mode: 11g
Web port: 80
Data port: 8888
Hostname: MYESP
Active Mode: STA (A4:CF:12:BF:90:57)
Connected to: Cisco5
Signal: 66%
IP Mode: DHCP
IP: 192.168.10.138
Gateway: 192.168.10.1
Mask: 255.255.255.0
DNS: 10.5.0.3
Disabled Mode: AP (A6:CF:12:BF:90:57)
Captive portal: Enabled
SSDP: Enabled
NetBios: Disabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: Marlin
M117 output: Enabled
Notifications: Disabled
Serial output: Enabled
Web socket  output: Enabled
TCP output: Enabled
FW version: 2.1.1.b8 ESP8266/8586

[ESP800]
FW version:2.1.1.b8 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81:192.168.10.138 # hostname:MYESP

[ESP410]plain
Cisco5	78	Secure
ASUS-X	34	Secure
PulsPass	8	Secure

vashilov, 2020-09-17:
>>how come after flashing you do not have index.html anymore ?
I try install FW many times. I have no index.html on screen, it is only cache.
But in brunch 2.1 FW version is 2.1.0.b39

Maybe i mistake))

luc-github, 2020-09-17:
no, your branch version is not 2.1.0.b39 - it is 2.1.1.b8 as show in [ESP800] (which is correct) where do you see the 2.1.0.b39 ?

vashilov, 2020-09-17:
I wrote

>>You use ESP3D-2.1-devt ?
Now I use it.

It ESP3D-2.1-devt branch.

vashilov, 2020-09-17:
Now info from 2.1 brach
[ESP420]plain
Chip ID: 12554327
CPU Frequency: 160Mhz
Free memory: 36.35 KB
SDK: 2.2.2-dev(38a443e)
Flash Size: 4.00 MB
Available Size for update: 535.87 KB(Ok)
Available Size for SPIFFS: 1.86 MB
Baud rate: 115200
Sleep mode: Modem
Channel: 11
Phy Mode: 11g
Web port: 80
Data port: 8888
Hostname: MYESP
Active Mode: STA (A4:CF:12:BF:90:57)
Connected to: Cisco5
Signal: 72%
IP Mode: DHCP
IP: 192.168.10.138
Gateway: 192.168.10.1
Mask: 255.255.255.0
DNS: 10.5.0.3
Disabled Mode: AP (A6:CF:12:BF:90:57)
Captive portal: Enabled
SSDP: Enabled
NetBios: Enabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: Marlin
M117 output: Enabled
Notifications: Disabled
Serial output: Enabled
Web socket  output: Enabled
TCP output: Enabled
FW version: 2.1.0.b39 ESP8266/8586
[ESP800]
FW version:2.1.0.b39 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81:192.168.10.138 # hostname:MYESP
[ESP410]plain
Cisco5	76	Secure
PulsPass	20	Secure
ASUS-X	34	Secure

vashilov, 2020-09-17:
Ok.
If I edit anywhith in config.h, example comment //#define DIRECT_PIN_FEATURE
I can not connect ti wifi ((

luc-github, 2020-09-17:
Sorry man but what are you doing exactly ? you looks changing branch version all the time
I think you mess up your source code and changed more code than what you describe - AUTHENTICATION_FEATURE and DIRECT_PIN_FEATURE have nothing related to WiFi neither WiFi Connection at boot time - so changing them have 0 impact on wifi connection 

1 - I suggest you to clean / delete all code 
2 - to download latest code from my repository branch 2.1
3 - change and do not copy from else where the AUTHENTICATION_FEATURE
4 - reflash using 4M with 2M SPIFFS
5 - do a [ESP444]RESET
6 - [ESP100]<your SSID>
7 -  [ESP101]<your SSID password>
8 - [ESP103]STA
9 - [ESP104]DHCP
10 - [ESP444]RESTART

vashilov, 2020-09-17:
>>you looks changing branch version all the time

I use 2.1 (2.1.0.b39).
You say me, it is old, I try use ESP3D-2.1-devt.

>>I suggest you to clean / delete all code

I already try it many times((

I do from 1 to 10:
1-4 ok
5-10 -> Incorrect Cmd

But [ESP420]plain is Ok

vashilov, 2020-09-17:
Any idea?
If it just foe me close disput.

Any people enable #define AUTHENTICATION_FEATURE?

luc-github, 2020-09-17:
`I use 2.1 (2.1.0.b39).` is not latest 2.1 branch version 
when enable #define AUTHENTICATION_FEATURE you need to provide admin password 

5 - [ESP444]RESET pwd=admin
6 - [ESP100] pwd=admin
7 - [ESP101] pwd=admin
8 - [ESP103]STA pwd=admin
9 - [ESP104]DHCP pwd=admin
10 - [ESP444]RESTART pwd=admin

vashilov, 2020-09-17:
Oh ok.
I mean AUTHENTICATION_FEATURE is password from web, no from cmd((
I wrote
>>P.S. AUTHENTICATION_FEATURE it is login and password to web interface?
ESP3D have password for WEB or no?

Wifi dont work any way for me!

[image]

[image]

vashilov, 2020-09-17:
I can edit ESP name, password, mode -> wifi work
If I comment or uncomment #define -> wiifi not connect

vashilov, 2020-09-17:
When ESP restarted I can connect to it for few miliseconds.
[image]

luc-github, 2020-09-17:
1 - you are not using correct source : https://github.com/luc-github/ESP3D/blob/2.1/esp3d/config.h#L22
2 - `I can edit ESP name, password, mode` where are you doing that ? you can only edit ESP Name in config.h

vashilov, 2020-09-17:
DEFAULT_AP_SSID
DEFAULT_ADMIN_PWD
DEFAULT_WIFI_MODE

vashilov, 2020-09-17:
If I open esp3d folder
In config.h I see #define FW_VERSION "2.1.1.b8"

If I click Download
[image]

I have 2.1.0.b39.
???

vashilov, 2020-09-17:
ESP3D have password for WEB or no?

luc-github, 2020-09-17:
DEFAULT_AP_SSID is for ESP3D AP mode not for Station mode
DEFAULT_ADMIN_PWD is for Authentication feature  (web - telnet - serial) not for AP neither STA mode

So your credential are wrong for sure and also you need to do a [ESP444]RESET to apply them or they are not

so if your version is 2.1.0.b39 it means the flash did not succeed  because it should be 2.1.1.b8 or you never closed arduino ide when you updated code to "2.1.1.b8 and ide still use old cache

luc-github, 2020-09-17:
also these define are after the important line :
`//Do not Edit after this line //////////////////////////////////////////////`

vashilov, 2020-09-17:
On 2.1.1.b8 AUTHENTICATION_FEATURE no work.
I update FW. And cmd use password.

[image]

>>DEFAULT_AP_SSID is for ESP3D AP mode not for Station mode
DEFAULT_ADMIN_PWD is for Authentication feature (web - telnet - serial) not for AP neither STA mode

I understand that.

vashilov, 2020-09-17:
>>//Do not Edit after this line

I understand that.
It for experement only!!

vashilov, 2020-09-17:
Please download zip file version 2.1 from web site.

What version you have in zip?

If I use git clone, I have 2.1.1.b8.
If download zip 2.1.0((

vashilov, 2020-09-18:
I tried using Visual Studio whith Platformio. All work fine.

Sorry man!!!

But try download zip file version 2.1 from web site.

closed 2020-09-18 by vashilov

luc-github, 2020-09-18:
[image]

vashilov, 2020-09-18:
Yes. I do this.
But not correct files inside.

My friend download zip, in him zip old FW too.

vashilov, 2020-09-18:
On mobile too.

[image]
[image]
[image]

luc-github, 2020-09-18:
Yes I have reproduced it - looks like github website link to release instead ot git - weird...