luc-github/ESP3D #508: [BUG] no printer config when using marlin 2.0.6

questionablemechanics, 2020-09-24:
**Describe the bug**
no printer configuration when using marlin 2.0

**To Reproduce**
Steps to reproduce the behavior:
Go to printers tab
Click on refresh
nothing except have label, value, help box
Expected behavior
list out full printer configuration

**M503 Report:**
`Changing monitoring state from "Detecting serial connection" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.0.6.1 (Sep 24 2020 10:44:11) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender 5 Plus EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:1
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:0
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M21
Recv: echo:SD card ok
Recv: ok
Send: M155 S2
Recv: ok
Send: M20
Recv: Begin file list
Recv: End file list
Recv: ok
Recv:  T:21.09 /0.00 B:20.94 /0.00 @:0 B@:0
Recv:  T:21.09 /0.00 B:21.09 /0.00 @:0 B@:0
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z800.00 E93.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E50.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 Z0.00
Recv: echo:  G29 W I0 J0 Z1.51708
Recv: echo:  G29 W I1 J0 Z1.63667
Recv: echo:  G29 W I2 J0 Z1.73208
Recv: echo:  G29 W I3 J0 Z1.81375
Recv: echo:  G29 W I4 J0 Z1.80625
Recv: echo:  G29 W I0 J1 Z1.82417
Recv: echo:  G29 W I1 J1 Z1.91458
Recv: echo:  G29 W I2 J1 Z1.99417
Recv: echo:  G29 W I3 J1 Z2.02250
Recv: echo:  G29 W I4 J1 Z2.03583
Recv: echo:  G29 W I0 J2 Z2.17667
Recv: echo:  G29 W I1 J2 Z2.30042
Recv: echo:  G29 W I2 J2 Z2.36542
Recv: echo:  G29 W I3 J2 Z2.37542
Recv: echo:  G29 W I4 J2 Z2.33792
Recv: echo:  G29 W I0 J3 Z2.53500
Recv: echo:  G29 W I1 J3 Z2.62292
Recv: echo:  G29 W I2 J3 Z2.66708
Recv: echo:  G29 W I3 J3 Z2.63208
Recv: echo:  G29 W I4 J3 Z2.60583
Recv: echo:  G29 W I0 J4 Z2.90000
Recv: echo:  G29 W I1 J4 Z2.98208
Recv: echo:  G29 W I2 J4 Z2.98875
Recv: echo:  G29 W I3 J4 Z2.93708
Recv: echo:  G29 W I4 J4 Z2.84875
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H205 B60 F0
Recv: echo:  M145 S1 H235 B80 F0
Recv: echo:; PID settings:
Recv: echo:  M301 P22.73 I1.65 D78.40
Recv: echo:  M304 P120.30 I9.77 D987.84
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S0
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-52.00 Y-20.00 Z-0.30
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X650 Y650 Z800
Recv: echo:  M906 T0 E800
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L350.00 U400.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1
Recv: ok
Recv:  T:20.94 /0.00 B:21.09 /0.00 @:0 B@:0
Recv:  T:21.09 /0.00 B:21.09 /0.00 @:0 B@:0
Recv:  T:20.94 /0.00 B:21.09 /0.00 @:0 B@:0
Recv:  T:20.94 /0.00 B:21.09 /0.00 @:0 B@:0
Recv:  T:20.94 /0.00 B:20.94 /0.00 @:0 B@:0
Recv:  T:21.09 /0.00 B:21.09 /0.00 @:0 B@:0
Recv:  T:20.94 /0.00 B:20.94 /0.00 @:0 B@:0`

**ESP3D Firmware:**
UI: 2.1b65
FW: 2.1.1.b8

**Target Firmware:**
Name: [ Marlin]
Version [2.0]

**Board used (please complete the following information):**
MCU: [esp01s and ESP32]
Name:[Doit devkit v1]
Flash size: [2M i think]

luc-github, 2020-09-24:
hi this is not readeable and this not M503 output, from which software this come from ? 

please 
what is output when send `M115` ?
what is output when send `M503` ?

questionablemechanics, 2020-09-24:
My apologies for the formatting. And thanks for the super fast reply.

**M503:**
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z800.00 E93.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E50.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S1 Z0.00
Recv: echo:  G29 W I0 J0 Z1.51708
Recv: echo:  G29 W I1 J0 Z1.63667
Recv: echo:  G29 W I2 J0 Z1.73208
Recv: echo:  G29 W I3 J0 Z1.81375
Recv: echo:  G29 W I4 J0 Z1.80625
Recv: echo:  G29 W I0 J1 Z1.82417
Recv: echo:  G29 W I1 J1 Z1.91458
Recv: echo:  G29 W I2 J1 Z1.99417
Recv: echo:  G29 W I3 J1 Z2.02250
Recv: echo:  G29 W I4 J1 Z2.03583
Recv: echo:  G29 W I0 J2 Z2.17667
Recv: echo:  G29 W I1 J2 Z2.30042
Recv: echo:  G29 W I2 J2 Z2.36542
Recv: echo:  G29 W I3 J2 Z2.37542
Recv: echo:  G29 W I4 J2 Z2.33792
Recv: echo:  G29 W I0 J3 Z2.53500
Recv: echo:  G29 W I1 J3 Z2.62292
Recv: echo:  G29 W I2 J3 Z2.66708
Recv: echo:  G29 W I3 J3 Z2.63208
Recv: echo:  G29 W I4 J3 Z2.60583
Recv: echo:  G29 W I0 J4 Z2.90000
Recv: echo:  G29 W I1 J4 Z2.98208
Recv: echo:  G29 W I2 J4 Z2.98875
Recv: echo:  G29 W I3 J4 Z2.93708
Recv: echo:  G29 W I4 J4 Z2.84875
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H205 B60 F0
Recv: echo:  M145 S1 H235 B80 F0
Recv: echo:; PID settings:
Recv: echo:  M301 P22.73 I1.65 D78.40
Recv: echo:  M304 P120.30 I9.77 D987.84
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S0
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-52.00 Y-20.00 Z-0.30
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X650 Y650 Z800
Recv: echo:  M906 T0 E800
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L350.00 U400.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1
Recv: ok

**M515:**
Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.6.1 (Sep 24 2020 10:44:11) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender 5 Plus EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:1
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:0
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M155 S2
Recv: ok

luc-github, 2020-09-24:
do you use webUI to send these commands ?

questionablemechanics, 2020-09-24:
No. This time was through Octoprint. Other times Pronterface.

luc-github, 2020-09-24:
can I have it from web UI ? I do not have 2.0.6.1 but 2.0.6.0
`Marlin 2.0.6 (Aug  9 2020 16:28:31)`

I want to know if there is any change or if it is a communication issue

questionablemechanics, 2020-09-24:
As I am unable to send codes, perhaps it's not a UI issue but a
communication issue? I've followed the Chris' Basement tutorial
https://www.youtube.com/watch?v=NGgzw-XayEo&ab_channel=ChrisRiley and have
updated the SERIAL_PORT s in Marlin. Alas...

luc-github, 2020-09-24:
yes if no code can be sent then it is not UI issue but Marlin Serial port configuration

questionablemechanics, 2020-09-24:
is it possible to have both a BTT TFT24 and the ESP3D? or only one or the
other? my current  serial ports are:

#define SERIAL_PORT -1

#define SERIAL_PORT_2 0

#define SERIAL_PORT_3 3

Any other configuration has rendered the TFT unusable.

luc-github, 2020-09-24:
Marlin only support 2 ports max not 3 
and you cannot share one port with 2 devices you willl get big mix up in communication

so if you connected your ESP to SERIAL_PORT_3  you won't have any communication as there is no SERIAL_PORT_3

questionablemechanics, 2020-09-24:
i appreciate the responses and your help!

luc-github, 2020-09-24:
the solutions are :
1 - do not use USB serial and use it for TFT  or esp
2 - use a BTT tft which do the bridge by connectiing ESP on TFT not on your board

questionablemechanics, 2020-09-24:
I suppose that could potentially mean connecting Octoprint via the TFT USB
and then the ESP via the board?

luc-github, 2020-09-24:
Hmm USB on BTT TFT is only for flash drive, no? it is not designed for USB communication

questionablemechanics, 2020-09-24:
there's the SD slot and then a standalone USB port:
https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/blob/master/TFT24-V1.1Manual.pdf
haven't ever used the USB port, but have used the SD slot for firmware

luc-github, 2020-09-24:
yes I did a PR to support BTT usb disk in webUI : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/648

TFT USB port is for USB disks

luc-github, 2020-09-28:
@questionablemechanics  any update ?

questionablemechanics, 2020-09-28:
have not had a chance to try it again through the TFT; waiting on a few
more parts and will update this thread once I try through the TFT.

closed 2020-10-23 by stale[bot]