luc-github/ESP3D #509: After last platformio update HardwareSerial.h causing problems

Drzet, 2020-09-24:
**Describe the bug**
In the last platformio update HardwareSerial.h UART definitions have changed and now include conditions
allowing to define pins and ports for specific UARTs. See code attached.
This has caused ESD32 to stop communicating with the board.
I was able to solve this by commenting all the "ifs" and "elses" but more graceful solution would be to use 
 **_define LPC_PINCFG_UART3_P4_28_**
However it is unclear where this definition should be placed.

`_For UART3:                      default is pin 46 and 47, PORT 0-00/01, FUNC 2
        define LPC_PINCFG_UART3_P0_25 to use pin  7 and  6, PORT 0-25/26, FUNC 3
        define LPC_PINCFG_UART3_P4_28 to use pin 82 and 85, PORT 4-28/29, FUNC 3_

and

__#if defined(LPC_PINCFG_UART3_P4_28)
        PinCfg.Funcnum = 3;
        PinCfg.Portnum = 4;
        PinCfg.Pinnum = 28;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 29;
        PINSEL_ConfigPin(&PinCfg);
      #elif defined(LPC_PINCFG_UART3_P0_25)
        PinCfg.Funcnum = 3;
        PinCfg.Portnum = 0;
       PinCfg.Pinnum = 25;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 26;
       PINSEL_ConfigPin(&PinCfg);
      #else
       PinCfg.Funcnum = 2;
       PinCfg.Portnum = 0;
       PinCfg.Pinnum = 0;
       PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 1;
        PINSEL_ConfigPin(&PinCfg);
      #endif`
**Target Firmware:**
 - Name: [e.g. Marlin]
 - Version [e.g. 2.0]

**Board used (please complete the following information):**
 - SKR1.4

luc-github, 2020-09-24:
Sorry - you are in ESP3D repository not Marlin - you should post this in Marlin repository

Drzet, 2020-09-24:
> Sorry - you are in ESP3D repository not Marlin - you should post this in Marlin repository

I understand that this is not an EPS32 bug but it is very unlikely to affect anything else but ESP32

luc-github, 2020-09-24:
Sorry I only handle esp3d not Marlin and this will affect any serial connected device like tft or BT devices or even esp board not using esp3d 

So if you issue is in Marlin and code modification is in Marlin and it is related to serial connection - solution is in Marlin not elsewhere, sorry

I suggest you to post your issue in Marlin repository

closed 2020-09-24 by Drzet